morrownr/8821au-20210708 #102: [Question] What is the reasonable speed?

burner1024, 2023-08-03:
Hi.
First off, thank you for the driver.
I'm using it with hostapd, providing a 5Ghz access point. Generally, `iwconfig` shows speed fluctuating in 250-400 Mb/s interval.
It's decent, but I wonder, is it near best expected speed, or there's a bottleneck somewhere that can be improved?

morrownr, 2023-08-03:
Hi @burner1024 

I have a speed comparison test that you can look at on the site Main Menu:

https://github.com/morrownr/USB-WiFi

Go to menu item 12.

While on the Main Menu, I recommend reading menu items 1 and 2. There is a lot of good information there for Linux users.

Let me know if you have any questions.

@morrownr

burner1024, 2023-08-03:
Great, thank you!

closed 2023-08-03 by burner1024