morrownr/8821au-20210708 #103: Ubuntu 22.04, kernel 6.2 compile error

burner1024, 2023-08-08:
```
/var/lib/dkms/rtl8812au/4.3.8.12175.20140902+dfsg/build/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/var/lib/dkms/rtl8812au/4.3.8.12175.20140902+dfsg/build/os_dep/osdep_service.c:2183:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2183 |         _rtw_memcpy(pnetdev->dev_addr, padapter->eeprompriv.mac_addr, ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/var/lib/dkms/rtl8812au/4.3.8.12175.20140902+dfsg/build/os_dep/osdep_service.c:807:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  807 | void _rtw_memcpy(void* dst, const void* src, u32 sz)
      |                  ~~~~~~^~~
/var/lib/dkms/rtl8812au/4.3.8.12175.20140902+dfsg/build/os_dep/osdep_service.c: In function ‘rtw_random32’:
/var/lib/dkms/rtl8812au/4.3.8.12175.20140902+dfsg/build/os_dep/osdep_service.c:2310:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘get_random_u32’? [-Werror=implicit-function-declaration]
 2310 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                get_random_u32
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:260: /var/lib/dkms/rtl8812au/4.3.8.12175.20140902+dfsg/build/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:2026: /var/lib/dkms/rtl8812au/4.3.8.12175.20140902+dfsg/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-26-generic'
make: *** [Makefile:1457: modules] Error 2
```

morrownr, 2023-08-08:
Hi @burner1024 

Need info. There is a pinned issue at the top that you should read.

While you are reading... I'll point out some things in what you posted:

/var/lib/dkms/rtl8812au/4.3.8.12175.20140902

So, the text showing errors is from a driver that is 9 years old. It is certainly not from the driver in this repo. Think about that while you read and compile the information that is needed.

After you post the needed info, let me invite you to the Main Menu here:

https://github.com/morrownr/USB-WiFi

Suggest you read menu items 1, 2 and 4. Realtek's out of kernel drivers are for programmers working on embedded projects, not us Ubuntu, Debian, Raspberry Pi OS, Arch users. Even finding the right driver, even understanding which driver is the right driver is very hard.

burner1024, 2023-08-08:
Sorry, Ubuntu must've pulled up the old dkms automatically. I had it then confused with the actual driver installed from this repo. After purging it, no more errors.

closed 2023-08-08 by burner1024