morrownr/8821au-20210708 #105: Request - Update for kernel 6.1?

Lightwel, 2023-08-18:
Hello  - request pls for new version to support  6.1 (running 6.1.0-11-amd64)

Tks

morrownr, 2023-08-19:
Hi @Lightwel 

This driver supports kernels well beyond 6.1. Tell me the problem you are seeing.

Lightwel, 2023-08-20:
Thanks for the fast response!  Simply put - device does not turn on.

I used the shell script to install and default blinking led. No blink.

lsusb shows the device and sudo dkms status shows the driver.  inxi -N and ip addr show nothing.

So I have done some digging.

#######
1    Here is the output of ./install-drivers.sh (run again after removing the device, to test)

richardh@richardh-SER:~/8821au-20210708$ sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 12/12 (in-use/total processing units)
: 29667524 (total system memory)
: 6.1.0-11-amd64 (kernel version)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh.........
Signing module /var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko
Cleaning build area...
The driver was built by dkms successfully.
: ---------------------------

8821au.ko:
Running module version sanity check.
- Original module
  - No original module exists within this kernel
- Installation
  - Installing to /lib/modules/6.1.0-11-amd64/updates/dkms/
depmod...
The driver was installed by dkms successfully.
: ---------------------------

Info: Upgrade this driver with the following commands as needed:
$ git pull
$ sudo sh install-driver.sh
Note: Upgrades to this driver should be performed before distro upgrades.
Note: Upgrades can be performed as often as you like.
Note: Work on this driver is continuous.

##########
2    Checking the driver is installed ..

richardh@richardh-SER:~$ sudo dkms status

rtl8821au/5.12.5.2, 6.1.0-11-amd64, x86_64: installed
richardh@richardh-SER:~$

###########
3    sudo dmesg before completing the shell setup script  shows

[  853.053706] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[  853.206769] usb 3-2: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 2.10
[  853.206788] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  853.206796] usb 3-2: Product: 802.11ac NIC
[  853.206803] usb 3-2: Manufacturer: Realtek
[  853.206809] usb 3-2: SerialNumber: 123456

##########
4    These are the selected options

options 8821au rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_dfs_region_domain=3 rtw_country_code=GB

##########
5    sudo dmesg after saving the options and rebooting shows
[    1.612734] usb 3-2: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 2.10
[    1.613627] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.614005] usb 3-2: Product: 802.11ac NIC
[    1.614389] usb 3-2: Manufacturer: Realtek
[    1.614764] usb 3-2: SerialNumber: 123456

##########
6    The system has an internal Mediatek wifi chip and I can see this being set up in

richardh@richardh-SER:~$ sudo dmesg | grep mt7921e
[    4.639112] mt7921e 0000:02:00.0: enabling device (0000 -> 0002)
[    4.664448] mt7921e 0000:02:00.0: ASIC revision: 79610010
[    4.749045] mt7921e 0000:02:00.0: firmware: direct-loading firmware mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
[    4.749729] mt7921e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20230117170855a
[    4.763399] mt7921e 0000:02:00.0: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
[    4.764466] mt7921e 0000:02:00.0: WM Firmware Version: ____010000, Build Time: 20230117170942
[    4.809855] mt7921e 0000:02:00.0: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
[    6.389021] mt7921e 0000:02:00.0 wlp2s0: renamed from wlan0

#########
7    But
richardh@richardh-SER:~$ sudo dmesg | grep rtl
[    5.442419] r8169 0000:01:00.0: firmware: direct-loading firmware rtl_nic/rtl8168h-2.fw

########
8    lsusb

Bus 003 Device 003: ID 0e8d:0608 MediaTek Inc. Wireless_Device
Bus 003 Device 002: ID 2357:0138 TP-Link 802.11ac NIC

##########
9    richardh@richardh-SER:~$ sudo dmesg | grep "usb 3-2"
[    1.460585] usb 3-2: new high-speed USB device number 2 using xhci_hcd
[    1.612734] usb 3-2: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 2.10
[    1.613627] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.614005] usb 3-2: Product: 802.11ac NIC
[    1.614389] usb 3-2: Manufacturer: Realtek
[    1.614764] usb 3-2: SerialNumber: 123456

That's all I can do ..

Tks

R

sunwire, 2023-08-20:
> Bus 003 Device 002: ID 2357:0138 TP-Link 802.11ac NIC

Well, I think you used the wrong driver. The correct one is https://github.com/morrownr/88x2bu-20210702

Lightwel, 2023-08-20:
Thanks for that - I believe I got the info that the 8812au driver was required from somewhere on the net which I cannot now find.

Anyway searching the dev id confirmed the ..bu driver which I then cloned and installed. 

Still no dice.

`ichardh@richardh-SER:~$ sudo dkms status
[sudo] password for richardh: 
rtl88x2bu/5.13.1, 6.1.0-11-amd64, x86_64: installed
richardh@richardh-SER:~$ inxi -N
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet
    driver: r8169
  Device-2: MEDIATEK MT7921K Wi-Fi 6E 80MHz driver: mt7921e
  Device-3: TP-Link 802.11ac NIC type: USB driver: N/A
richardh@richardh-SER:~$ sudo dmesg | grep "usb 3-2"
[    1.960390] usb 3-2: new high-speed USB device number 2 using xhci_hcd
[    2.851259] usb 3-2: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 2.10
[    2.851930] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.852612] usb 3-2: Product: 802.11ac NIC
[    2.853263] usb 3-2: Manufacturer: Realtek
[    2.853911] usb 3-2: SerialNumber: 123456
`

I know the adapter works because I have no issues in Kubuntu.

??

Tks
R

Lightwel, 2023-08-20:
SOLVED

I installed the same options as before 

options 8821au rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_dfs_region_domain=3 rtw_country_code=GB

But it worked when I removed rtw_country_code=GB

Tks
R

yuralamov, 2023-09-13:
6.5.1-1-MANJARO - ok

closed 2023-11-06 by morrownr