morrownr/8821au-20210708 #107: 8821au installs fine, module loads, but no interface?

hjheins, 2023-08-29:
Hello,
I installed your package with no issues whatsoever.
./install-driver.sh worked fine, and without errors. However, after reboot, the module doesn't load automatically.
Also, when loading the module with "modprobe 8821au" I get the following in dmesg:

[code]
[ 4712.483902] usb 3-2: new high-speed USB device number 2 using xhci_hcd
[ 4712.630199] usb 3-2: New USB device found, idVendor=2604, idProduct=0012, bcdDevice= 0.00
[ 4712.630213] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4712.630218] usb 3-2: Product: 802.11n NIC
[ 4712.630222] usb 3-2: Manufacturer: Realtek
[ 4712.630225] usb 3-2: SerialNumber: 123456
[ 4996.813913] usb 3-2: USB disconnect, device number 2
[ 5003.957132] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[ 5004.100024] usb 3-2: New USB device found, idVendor=2604, idProduct=0012, bcdDevice= 0.00
[ 5004.100038] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5004.100043] usb 3-2: Product: 802.11n NIC
[ 5004.100047] usb 3-2: Manufacturer: Realtek
[ 5004.100051] usb 3-2: SerialNumber: 123456
[ 5013.091651] RTW: module init start
[ 5013.091658] RTW: rtl8821au v5.12.5.2-0-g70054197b.20210708_COEX20190509-6d6f
[ 5013.091662] RTW: rtl8821au BT-Coex version = COEX20190509-6d6f
[ 5013.091757] usbcore: registered new interface driver rtl8821au
[ 5013.091760] RTW: module init ret=0
[/code]

However, when I check with "rfkill" or "iwconfig", there is no interface?
My usb module is a Tenda U12, which, according to what I found in the documentation, is supported (dev and usb ID match).

The system I tested on:
- odroid N with Ubuntu 20
- arch Linux on an Asus Zenbook
I get the same result for both systems.

Not sure what to do next? How could I get this working?
Thanks,

Hendrik-Jan

morrownr, 2023-08-30:
Hi @hjheins 

There is a file called supported-device-IDs in the driver directory. Read it and check to see if you are using the correct driver.

If not, paste the result of:

$ lsusb

hjheins, 2023-08-30:
Hello @morrownr,
In the file "supported-device-IDs" my usb device is not mentioned.
However in the referred file: "usb_intf.c" it is mentioned on line 190, in the section for the 8812A.

The driver that is being built is: "8821au". Not sure if this is correct?

lsusb gives:
[code]
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2604:0012 Tenda U12
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 002: ID 0408:30c0 Quanta Computer, Inc. USB2.0 HD UVC WebCam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[/code]

thanks.

hjheins, 2023-08-30:
Ok, i Found it.
That was officially a stupid move:
I installed the 8821au driver from your repository, which is NOT for the 8812au (see the number switch at the back.)
I just found teh 8812au repo, installed it, and now it is working as designed.

Thank you very much for your work!
Ticket will be closed.

closed 2023-08-30 by hjheins

morrownr, 2023-08-30:
Good.

Let me point out the site Main Menu as it may have some info that is of use to you:

https://github.com/morrownr/USB-WiFi