morrownr/8821au-20210708 #108: [6.5.3] Pointer type issue.

userwiths, 2023-09-17:
Had postponed system update of my PC for a few weeks (maybe months) did:
> \> `pacman -Syu` (updated linux kernel to 6.5.3).
> \> `reboot`
> \> `./install-driver.sh` (install this driver)
> \> Get error about `Initialization of ... from incompatible pointer type ...`

After downgrading to 6.3.5 the driver installed without any issues.

Exact line throwing the error: https://github.com/morrownr/8821au-20210708/blob/19660ee3f67125d2ef31b626f9fff946bd3e8d01/os_dep/linux/ioctl_cfg80211.c#L10545

Exact Output Error:
```
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 16/16 (in-use/total processing units)
: 31956684 (total system memory)
: 6.5.3-arch1-1 (kernel version)
: gcc (GCC) 13.2.1 20230801
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The non-dkms installation routines are in use.
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.5.3-arch1-1/build M=/root/driver/8821au-20210708  modules
make[1]: Entering directory '/usr/lib/modules/6.5.3-arch1-1/build'
  CC [M]  /root/driver/8821au-20210708/core/rtw_cmd.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_security.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_debug.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_io.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_ioctl_query.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_ioctl_set.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_ieee80211.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_mlme.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_mlme_ext.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_mi.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_wlan_util.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_vht.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_pwrctrl.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_rf.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_chplan.o
  CC [M]  /root/driver/8821au-20210708/core/monitor/rtw_radiotap.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_recv.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_ap.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_sta_mgt.o
  CC [M]  /root/driver/8821au-20210708/core/wds/rtw_wds.o
  CC [M]  /root/driver/8821au-20210708/core/mesh/rtw_mesh.o
  CC [M]  /root/driver/8821au-20210708/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/driver/8821au-20210708/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_xmit.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_p2p.o
  CC [M]  /root/driver/8821au-20210708/core/rtw_rson.o
In file included from ./include/linux/kernel.h:30,
                 from ./arch/x86/include/asm/percpu.h:27,
                 from ./arch/x86/include/asm/nospec-branch.h:14,
                 from ./arch/x86/include/asm/paravirt_types.h:27,
                 from ./arch/x86/include/asm/ptrace.h:97,
                 from ./arch/x86/include/asm/math_emu.h:5,
                 from ./arch/x86/include/asm/processor.h:13,
                 from ./arch/x86/include/asm/timex.h:5,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /root/driver/8821au-20210708/include/basic_types.h:76,
                 from /root/driver/8821au-20210708/include/drv_types.h:26,
[... 110 lines cut ...]
  CC [M]  /root/driver/8821au-20210708/hal/hal_mp.o
  CC [M]  /root/driver/8821au-20210708/hal/hal_mcc.o
  CC [M]  /root/driver/8821au-20210708/hal/hal_hci/hal_usb.o
  CC [M]  /root/driver/8821au-20210708/hal/led/hal_led.o
  CC [M]  /root/driver/8821au-20210708/hal/led/hal_usb_led.o
  CC [M]  /root/driver/8821au-20210708/hal/HalPwrSeqCmd.o
  CC [M]  /root/driver/8821au-20210708/hal/rtl8812a/Hal8812PwrSeq.o
  CC [M]  /root/driver/8821au-20210708/hal/rtl8812a/Hal8821APwrSeq.o
  CC [M]  /root/driver/8821au-20210708/hal/rtl8812a/rtl8812a_xmit.o
  CC [M]  /root/driver/8821au-20210708/hal/rtl8812a/rtl8812a_sreset.o
  CC [M]  /root/driver/8821au-20210708/hal/rtl8812a/rtl8812a_hal_init.o
  CC [M]  /root/driver/8821au-20210708/hal/rtl8812a/rtl8812a_phycfg.o
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /root/driver/8821au-20210708/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/lib/modules/6.5.3-arch1-1/build/Makefile:2032: /root/driver/8821au-20210708] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.5.3-arch1-1/build'
make: *** [Makefile:2501: modules] Error 2
An error occurred:  2
Please report this error.
Please copy all screen output and paste it into the problem report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

userwiths, 2023-09-17:
`iw dev` output: 
```
hy#0
	Interface wlan0
		ifindex 8
		wdev 0x1
		addr 28:87:ba:cc:fd:39
		ssid Space
		type managed
		channel 8 (2447 MHz), width: 20 MHz, center1: 2447 MHz
		txpower 16.00 dBm
```

`iw reg get`:
```
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

`lsusb` output:
```
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 002: ID 26ce:01a2 ASRock LED Controller
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 3434:0331 Keychron Keychron V3
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046d:082d Logitech, Inc. HD Pro Webcam C920
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 046d:c548 Logitech, Inc. Logi Bolt Receiver
Bus 001 Device 003: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

morrownr, 2023-09-17:
Hi @userwiths 

This is happening because the driver has not been updated on your computer for quite a while. I have been working on install-driver.sh to add some new features and to provide information that would prevent this. I'll update this driver as able but for now, let me post what I am adding about updating the driver:

-----

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates can be performed as often as you like. It is
      recommended to update at least every 2 months.
Note: Work on this driver, like the Linux kernel, is continuous.

-----

Go ahead and try the above commands and see if you are back in business.

userwiths, 2023-09-18:
Oh, damn. I feel a little stupid now.

You are correct, sorry for the inconvenience and thanks for the assistance, will make sure to update the driver regularly from now on.

closed 2023-09-18 by userwiths

morrownr, 2023-09-18:
Hi @userwiths 

Don't worry. These Realtek out-of-kernel drivers are not made for us regular users. They seem to be made for programmers working on embedded projects. I've done the best I can to help and I continue that work but there is an alternative that you may want to know about going into the future:

The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Recommend you read menu items 1 and 2 to understand the current state of the Linux usb wifi market and see many adapters that use in-kernel drivers. AXE3000 usb wifi adapters are available and the WiFi 7 usb driver for the Mediatek mt7925 is going into the kernel at this time.