morrownr/8821au-20210708 #110: how to compile as an external kernel module?

erophey7, 2023-10-16:
I tried this instruction: https://www.kernel.org/doc/Documentation/kbuild/modules.txt, but it didn’t work:
`erophey7@compudahter-kubuntu:~/eblya/8821au-20210708$ make -C $HOME/eblya/android_kernel_xiaomi_lahaina/out M=$PWD ARCH=arm64 CC=clang CLANG_TRIPLE=aarch64-linux-gnu- LLVM=1 LLVM_IAS=1 CROSS_COMPILE=$TC_DIR/bin/llvm-
make: Entering directory '/home/erophey7/eblya/android_kernel_xiaomi_lahaina/out'
  Building modules, stage 2.
  MODPOST 0 modules
make: Leaving directory '/home/erophey7/eblya/android_kernel_xiaomi_lahaina/out'
`

morrownr, 2023-10-24:
Hi @erophey7 

I have a lot of patches nearly ready to go it and one may help you out. Keep an eye out for newly merged patches.

morrownr, 2023-11-06:
I've now merged a large set of patches that might help.

closed 2024-03-10 by erophey7