morrownr/8821au-20210708 #113: Kernel 5.15.95-xanmod1 : modprobe: ERROR: could not insert '8821au': Exec format error

RJVB, 2023-10-24:
I built and installed a 5.15.95 xanmod1 kernel on a system where I had already built and installed working versions of this driver for 5.6 kernels with the Con Kolivas and with the Liquorix patch sets.

The build completes fine but I get the following output from modinfo and modprobe:

```
> modinfo 8821au
filename:       /lib/modules/5.15.95-xanmod1-westmere-rjvb+/updates/dkms/8821au.ko
version:        v5.12.5.2-0-g70054197b.20210708_COEX20190509-6d6f
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     EFD6C9907EE82CEC4897226
alias:          usb:v20F4p804Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0120d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p011Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p011Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v3823p6249d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4007d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p029Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0846p9052d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB32d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p0242d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E66p0023d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3318d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3314d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04BBp0953d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pB611d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0823d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp0820d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8822d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0821d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0811d*dc*dsc*dp*ic*isc*ip*in*
depends:        cfg80211
retpoline:      Y
name:           8821au
vermagic:       5.10.113-xanmod1 SMP mod_unload 
parm:           rtw_wireless_mode:int
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           rtw_lps_level:The default LPS level (int)
parm:           rtw_lps_chk_by_tp:int
parm:           rtw_max_bss_cnt:int
parm:           rtw_usb_rxagg_mode:int
parm:           rtw_dynamic_agg_enable:int
parm:           rtw_drv_log_level:set log level when insert driver module, default log level is _DRV_INFO_ = 4 (uint)
parm:           rtw_tx_aclt_flags:device TX AC queue packet lifetime control flags (uint)
parm:           rtw_tx_aclt_conf_default:device TX AC queue lifetime config for default status (array of uint)
parm:           rtw_tx_aclt_conf_ap_m2u:device TX AC queue lifetime config for AP mode M2U status (array of uint)
parm:           rtw_tx_bw_mode:The max tx bw for 2.4G and 5G. format is the same as rtw_bw_mode (uint)
parm:           rtw_rx_ampdu_sz_limit_1ss:RX AMPDU size limit for 1SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rx_ampdu_sz_limit_2ss:RX AMPDU size limit for 2SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rx_ampdu_sz_limit_3ss:RX AMPDU size limit for 3SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rx_ampdu_sz_limit_4ss:RX AMPDU size limit for 4SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_stbc_cap:int
parm:           rtw_vht_enable:int
parm:           rtw_vht_24g_enable:int
parm:           rtw_vht_rx_mcs_map:VHT RX MCS map (uint)
parm:           rtw_trx_path_bmp:int
parm:           rtw_tx_path_lmt:int
parm:           rtw_rx_path_lmt:int
parm:           rtw_tx_nss:int
parm:           rtw_rx_nss:int
parm:           rtw_country_code:The default country code (in alpha2) (charp)
parm:           rtw_channel_plan:The default chplan ID when rtw_alpha2 is not specified or valid (int)
parm:           rtw_excl_chs:exclusive channel array (array of uint)
[... 76 lines cut ...]
parm:           rtw_target_tx_pwr_5g_d:5G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_antenna_gain:Antenna gain in mBi. 0x7FFF: unspecifed (int)
parm:           rtw_tsf_update_pause_factor:num of bcn intervals to stay TSF update pause status (int)
parm:           rtw_tsf_update_restore_factor:num of bcn intervals to stay TSF update restore status (int)
parm:           rtw_phy_file_path:The path of phy parameter (charp)
parm:           rtw_load_phy_file:PHY File Bit Map (int)
parm:           rtw_decrypt_phy_file:Enable Decrypt PHY File (int)
parm:           rtw_recvbuf_nr:Preallocated number of struct recv_buf (int)
parm:           rtw_phydm_ability:uint
parm:           rtw_halrf_ability:uint
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_ch_switch_offload:int
parm:           rtw_en_tdls:int
parm:           rtw_en_dyn_rrsr:int
parm:           rtw_rrsr_value:int
parm:           rtw_scan_interval_thr:Threshold used to judge if scan request comes from scan UI, unit is ms. (uint)

> modprobe -v 8821au
modprobe: ERROR: could not insert '8821au': Exec format error
insmod /lib/modules/5.15.95-xanmod1-westmere-rjvb+/updates/dkms/8821au.ko rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 
```

I see no unexpected differences in the modinfo output with the output from a kernel where the module does work, so what's this about an exec format error?

morrownr, 2023-10-24:
Hi @RJVB 

> so what's this about an exec format error?

No idea. Did you use install-driver.sh for the installation? Run it again and post the screen output.

RJVB, 2023-10-24:
>No idea. Did you use install-driver.sh for the installation? Run it again and post the screen output.

For the initial installation in one of the kernels where the driver works, yes. All other builds have been via dkms.

morrownr, 2023-10-24:
Try running install-driver.sh and then copy what it puts on the screen to here.

That will give me a better idea of your setup. FYI: the script uses dkms if installed and does a manual installation if it is not installed. No magic here, just automation along with a LOT of error checking.

RJVB, 2023-10-24:
Will do when I get access to the machine again!

RJVB, 2023-10-25:
>That will give me a better idea of your setup. FYI: the script uses dkms if installed and does a manual installation if it is not installed. No magic here, just automation along with a LOT of error checking.

It seems something must have gone wrong during the initial build. I don't see anything of use in the output you requested but the module now loads and the dongle works.

So thanks for making me do this :)

morrownr, 2023-10-25:
You are welcome.

@morrownr