morrownr/8821au-20210708 #116: Trouble Switching to Monitor Mode

VHS2, 2023-12-16:
I want to start by thanking morrownr for his extensive work on this project, and everyone else who contributed.

 I've been doing a decent job following the instructions to a T but I've seem to hit a road block. Usually I can go back, re-read, and find a solution but this time I'm stuck for real.

I was able to install the driver successfully.
Switching to monitor mode is where the problems start.

I try running **sudo ./start-mon.sh wlan0** it says:  

The following processes have been stopped:

    PID Name
    598 NetworkManager
 139803 wpa_supplicant
 
 it then gives me the option to set my txpower and i set it to 1600 as prompted in the instructions. After i set it to 1600 it reads:
 
  --------------------------------
    start-mon.sh 20230305
 --------------------------------
    WiFi Interface:
             wlan0
 --------------------------------
    name  -  wlan0mon
    type  -  managed
    state -  DORMANT
    addr  -  c2:2a:0d:b8:dc:f9
    chan  -  
    txpw  -  -100.00 dBm
 --------------------------------

 DORMANT = up but inactive.

 Ready for Monitor Mode use.

 
 
 Then i run **iw dev** it says: 
 

phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr d6:c7:82:b0:27:8e
                type managed
                txpower -100.00 dBm
                                      

as you see the type is still set to managed and the txpower hasn't changed.

I also tried running the command: **sudo ./stop-procs.sh wlan0**
and this is the result: 

 : stop-procs.sh v20230220

  The following processes have been stopped:

    PID Name
    598 NetworkManager
 139803 wpa_supplicant

  Press 'Enter' when you are ready to restart the above processes.    

  The following processes have been restarted:

    PID Name
    598 NetworkManager
 139803 wpa_supplicant

Doing this STILL doesn't get it into monitor mode. I Also tried all the steps/prompts in Monitor_Mode.md to no avail.

Any recommendations will be greatly appreciated!

morrownr, 2023-12-18:
Hi @VHS2 

Let try the manual way to get into monitor mode. Reboot, then enter the following commands. Copy all and post here:

```
Check the wifi interface name and mode

iw dev

Take the interface down

sudo ip link set <wlan0> down

Set monitor mode

sudo iw <wlan0> set monitor none

Bring the interface up

sudo ip link set <wlan0> up

Verify the mode has changed

iw dev
```

Also, provide your hardware and distro.

@VHS2

VHS2, 2023-12-18:
> Hi @VHS2
>
> Let try the manual way to get into monitor mode. Reboot, then enter the following commands. Copy all and post here:
>
> ```
> Check the wifi interface name and mode
>
> iw dev
>
> Take the interface [...]

thanks for replying! Ok so here are the results:

┌──(user1㉿kali)-[~]
└─$ iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr da:28:6a:91:10:69
                type managed
                txpower -100.00 dBm
                                                                                                  
┌──(user1㉿kali)-[~]
└─$ sudo ip link set wlan0 down   
[sudo] password for user1: 
                                                                                                  
┌──(user1㉿kali)-[~]
└─$ sudo iw wlan0 set monitor none
command failed: Operation not permitted (-1)
                                                                                                  
┌──(user1㉿kali)-[~]
└─$ sudo ip link set wlan0 up     
                                                                                                  
┌──(user1㉿kali)-[~]
└─$ iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr da:28:6a:91:10:69
                type managed
                txpower -100.00 dBm
                                                                                                  
┌──(user1㉿kali)-[~]
└─$ 

MY HARDWARE:

Intel Core i3-4130 CPU @ 3.40GHz
64bits

MY DISTRO:

Im running Kali Linux on a VM
When i run ─$ cat /etc/os-release

it says:  ID_LIKE - DEBIAN 

and i successfully used the Option for Ubuntu (all official flavors)  to install the necessary packages:
sudo apt install -y build-essential dkms git iw

(I Believe the Debian option was giving me a hard time)

morrownr, 2023-12-18:
>
``` ┌──(user1㉿kali)-[~]
└─$ sudo iw wlan0 set monitor none
command failed: Operation not permitted (-1)
```

That is not what we want to see. We forgot to stop the interfering processes. Let me give you another list to try:

```
sudo ./stop-procs.sh wlan0

When it gets to `Press Enter when you are ready to restart the above processes.' open another terminal and run the following:

Check the wifi interface name and mode

iw dev

ip a

Take the interface down

sudo ip link set <wlan0> down

ip a

Set monitor mode

sudo iw <wlan0> set monitor none

Bring the interface up

sudo ip link set <wlan0> up

Verify the mode has changed

iw dev

```

Post all of the results.

> Im running Kali Linux on a VM..

Let me rant for a moment. Kali in a VM is a recipe for frustration. My recommendation is to find an older computer that you can buy cheap or a RasPi4B or something where you can run Linux on bare metal. Rant off.

@morrownr

VHS2, 2023-12-18:
> > 
> 
> ```
> └─$ sudo iw wlan0 set monitor none
> command failed: Operation not permitted (-1)
> ```
> 
> That is not what we want to see. We forgot to stop the interfering processes. Let me give you another list to try:

> Post all of the results.
> 
> > Im running Kali Linux on a VM..
> 
> Let me rant for a moment. Kali in a VM is a recipe for frustration. My recommendation is to find an older computer that you can buy cheap or a RasPi4B or something where you can run Linux on bare metal. Rant off.
> 
> @morrownr

┌──(user1㉿kali)-[~]
└─$ iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 0e:14:19:03:12:4d
                type managed
                txpower -100.00 dBm
                                                                           
┌──(user1㉿kali)-[~]
└─$ ip a  
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 08:00:27:68:98:c8 brd ff:ff:ff:ff:ff:ff
    inet 10.0.2.15/24 brd 10.0.2.255 scope global dynamic noprefixroute eth0
       valid_lft 75586sec preferred_lft 75586sec
    inet6 fe80::a00:27ff:fe68:98c8/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state DORMANT group default qlen 1000
    link/ether 0e:14:19:03:12:4d brd ff:ff:ff:ff:ff:ff permaddr b4:b0:24:80:20:82
                                                                           
┌──(user1㉿kali)-[~]
└─$ sudo ip link set wlan0 down
[sudo] password for user1: 
                                                                           
┌──(user1㉿kali)-[~]
└─$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 08:00:27:68:91:c8 brd ff:ff:ff:ff:ff:ff
    inet 10.0.2.15/24 brd 10.0.3.255 scope global dynamic noprefixroute eth0
       valid_lft 75551sec preferred_lft 75551sec
    inet6 fe80::a00:17ff:fe67:88c8/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 0e:18:13:03:17:4d brd ff:ff:ff:ff:ff:ff permaddr b2:b0:44:80:20:81                                                                     
                                                                           
┌──(user1㉿kali)-[~]
└─$ sudo iw wlan0 set monitor none
command failed: Operation not permitted (-1)
                                                                           
┌──(user1㉿kali)-[~]
└─$ sudo ip link set wlan0 up     
                                                                           
┌──(user1㉿kali)-[~]
└─$ iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 0e:14:19:03:12:4d
                type managed
                txpower -100.00 dBm
                                                                           
┌──(user1㉿kali)-[~]
└─$ 

So I'm getting the same error in the new terminal:

 └─$ sudo iw wlan0 set monitor none
command failed: Operation not permitted (-1) 

.

(Reply to Rant)

And YES, I am now witnessing the frustration first hand LOL
I'll definitely consider getting the RasPi if it's going to make my life easier!

Whats wrong with the cheap old computer I'm using now? Should I just run Linux as the main OS? Or is that not ideal if this is my only CPU at the moment?

 In the meantime will it make sense for me to get a cheap USB WiFi adapter that is supported with Linux in-kernel drivers? Or is it a good chance that i will still have the same issue being that I'm on a VM?

VHS2, 2023-12-18:
Ok so it looks like i found a fix, 

I closed Linux, 
Open vmware machine settings > network > adapter 2 > enable network adapter>   choose host only adapter > apply

i was finally able to get it to change to monitor mode but firefox wont load any pages so there's another problem I'll have fun with.

Thanks for your help! @morrownr

closed 2023-12-18 by VHS2

morrownr, 2023-12-18:
Something you might want to research is multiboot.

What is multiboot in Linux?

Multi-booting is the process of installing and running more than one operating system on a single computer. It can be useful for testing, experimenting, or switching between different environments.

It allows you to have more than one os installed on a computer. On boot, a screen appears that allows you to boot into the os you want to use. All of the installers on mainstream Linux distros allow you to set this up during installation. The installer will reduce the size of the Windows or other os partition so as to make room for a fully separate installation of Linux. No VM is needed in this setup. Guides are available for many distros.

@morrownr