morrownr/8821au-20210708 #117: System hangs on Debian 12 with Linux 6.1.0-15-amd64

insel-maz, 2023-12-17:
I have problems with the wifi driver on Debian 12 (bookworm), KDE.

The last version of Debian that works for me is:
Linux 6.1.0-14-amd64 (6.1.64-1)
The system does not connect to the wifi and the system hangs with this version:
Linux 6.1.0-15-amd64 (6.1.66-1)

The device is a tp-link AC600 Archer T600U Plus Antenna.

Output of `sudo dmesg -w`:

```
[  242.748008] INFO: task wpa_supplicant:884 blocked for more than 120 seconds.
[  242.748015]       Tainted: G           OE      6.1.0-15-amd64 #1 Debian 6.1.66-1
[  242.748017] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  242.748021] task:wpa_supplicant  state:D stack:0     pid:884   ppid:1      flags:0x00004002
[  242.748025] Call Trace:
[  242.748027]  <TASK>
[  242.748029]  __schedule+0x34d/0x9e0
[  242.748036]  schedule+0x5a/0xd0
[  242.748037]  schedule_preempt_disabled+0x11/0x20
[  242.748039]  __mutex_lock.constprop.0+0x399/0x700
[  242.748042]  ? __nla_reserve+0x38/0x50
[  242.748046]  nl80211_send_iface+0x25b/0x970 [cfg80211]
[  242.748101]  ? kmalloc_reserve+0x4b/0xa0
[  242.748105]  ? __alloc_skb+0xd9/0x1a0
[  242.748107]  nl80211_get_interface+0x4b/0xa0 [cfg80211]
[  242.748142]  genl_family_rcv_msg_doit+0x100/0x160
[  242.748148]  genl_rcv_msg+0x122/0x250
[  242.748150]  ? nl80211_dump_interface+0x1e0/0x1e0 [cfg80211]
[  242.748185]  ? nl80211_send_iface+0x970/0x970 [cfg80211]
[  242.748219]  ? genl_start+0x160/0x160
[  242.748222]  netlink_rcv_skb+0x51/0x100
[  242.748225]  genl_rcv+0x24/0x40
[  242.748228]  netlink_unicast+0x24a/0x390
[  242.748230]  netlink_sendmsg+0x250/0x4c0
[  242.748233]  __sock_sendmsg+0x5f/0x70
[  242.748236]  ____sys_sendmsg+0x277/0x2f0
[  242.748238]  ? copy_msghdr_from_user+0x7d/0xc0
[  242.748241]  ___sys_sendmsg+0x9a/0xe0
[  242.748246]  __sys_sendmsg+0x76/0xc0
[  242.748249]  do_syscall_64+0x5b/0xc0
[  242.748252]  ? __sys_setsockopt+0xf8/0x200
[  242.748255]  ? exit_to_user_mode_prepare+0x40/0x1e0
[  242.748258]  ? syscall_exit_to_user_mode+0x27/0x40
[  242.748260]  ? do_syscall_64+0x67/0xc0
[  242.748263]  ? syscall_exit_to_user_mode+0x27/0x40
[  242.748265]  ? do_syscall_64+0x67/0xc0
[  242.748267]  ? do_syscall_64+0x67/0xc0
[  242.748269]  entry_SYSCALL_64_after_hwframe+0x64/0xce
[  242.748273] RIP: 0033:0x7f7150f29930
[  242.748275] RSP: 002b:00007fffe61c7c68 EFLAGS: 00000202 ORIG_RAX: 000000000000002e
[  242.748277] RAX: ffffffffffffffda RBX: 000055b96d93f8e0 RCX: 00007f7150f29930
[  242.748278] RDX: 0000000000000000 RSI: 00007fffe61c7ca0 RDI: 0000000000000006
[  242.748279] RBP: 000055b96d9e92f0 R08: 0000000000000004 R09: 00007f7150ff1fb0
[  242.748280] R10: 00007fffe61c7d84 R11: 0000000000000202 R12: 000055b96d93fbc0
[  242.748282] R13: 00007fffe61c7ca0 R14: 000055b96c319a40 R15: 00007fffe61c7d84
[  242.748284]  </TASK>
[  363.579932] INFO: task kworker/0:0:7 blocked for more than 120 seconds.
[  363.579940]       Tainted: G           OE      6.1.0-15-amd64 #1 Debian 6.1.66-1
[  363.579942] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  363.579943] task:kworker/0:0     state:D stack:0     pid:7     ppid:2      flags:0x00004000
[  363.579948] Workqueue: events_power_efficient reg_check_chans_work [cfg80211]
[  363.580019] Call Trace:
[  363.580020]  <TASK>
[  363.580023]  __schedule+0x34d/0x9e0
[  363.580029]  schedule+0x5a/0xd0
[  363.580031]  schedule_preempt_disabled+0x11/0x20
[  363.580033]  __mutex_lock.constprop.0+0x399/0x700
[  363.580037]  reg_check_chans_work+0x2d/0x5e0 [cfg80211]
[... 254 lines cut ...]
[  363.580803]  __mutex_lock.constprop.0+0x399/0x700
[  363.580806]  __netlink_dump_start+0xc1/0x2f0
[  363.580808]  ? rtnl_fill_ifinfo+0x14c0/0x14c0
[  363.580810]  rtnetlink_rcv_msg+0x273/0x3b0
[  363.580812]  ? rtnl_fill_ifinfo+0x14c0/0x14c0
[  363.580814]  ? rtnl_calcit.isra.0+0x140/0x140
[  363.580816]  netlink_rcv_skb+0x51/0x100
[  363.580819]  netlink_unicast+0x24a/0x390
[  363.580822]  netlink_sendmsg+0x250/0x4c0
[  363.580825]  __sock_sendmsg+0x5f/0x70
[  363.580827]  __sys_sendto+0x11c/0x170
[  363.580829]  ? __sys_setsockopt+0x12e/0x200
[  363.580832]  __x64_sys_sendto+0x20/0x30
[  363.580834]  do_syscall_64+0x5b/0xc0
[  363.580837]  ? do_syscall_64+0x67/0xc0
[  363.580839]  entry_SYSCALL_64_after_hwframe+0x64/0xce
[  363.580842] RIP: 0033:0x7f64d9b298c4
[  363.580843] RSP: 002b:00007f64bcdfdd80 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
[  363.580845] RAX: ffffffffffffffda RBX: 0000000000000015 RCX: 00007f64d9b298c4
[  363.580846] RDX: 0000000000000020 RSI: 00007f64bcdfde20 RDI: 0000000000000015
[  363.580847] RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000000
[  363.580848] R10: 0000000000000000 R11: 0000000000000246 R12: 00007f64bcdfdee8
[  363.580849] R13: 00007f64bcdfddf0 R14: 0000000000000001 R15: 0000563febe80a69
[  363.580851]  </TASK>
```

morrownr, 2023-12-17:
Hi @insel-maz 

This driver is not the cause. See: https://github.com/morrownr/USB-WiFi/issues/349

insel-maz, 2024-01-06:
Hi, with Debian 12 kernel version 6.1.0-17-amd64 (6.1.69-1) the driver works well again.

Thank you for your time. May I buy you a coffee?

morrownr, 2024-01-06:
> Thank you for your time.

You are welcome. Glad you are up and running again.

> May I buy you a coffee?

I appreciate the offer but the thank you is more than enough.

Take care.

@morrownr

closed 2024-01-07 by insel-maz