morrownr/8821au-20210708 #118: Build works properly in Debian testing

MrReplikant, 2023-12-26:
This is just here to let you know that the drive works as expected in Linux Kernel 6.5, in Debian Testing.

morrownr, 2023-12-26:
Hi @MrReplikant 

Thanks for the report. It should be working with kernel 6.7 also.

TDYK, 2024-01-26:
Hi. @morrownr 

I have an issue with 6.7 kernel in Debian GNU/Linux trixie/sid:

noroot@workstation:~/bin/8821au-20210708$ git pull
Already up to date.

dkms build -k 6.7.2-1-liquorix-amd64 -m rtl8821au -v 5.12.5.2
The kernel is built without module signing facility, modules won't be signed

Building module:
Cleaning build area...
./dkms-make.sh.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.7.2-1-liquorix-amd64 (x86_64)
Consult /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log for more information.

Tried with different gcc versions also

Attached files:
[make.log](https://github.com/morrownr/8821au-20210708/files/14068380/make.log)
[info.txt](https://github.com/morrownr/8821au-20210708/files/14068383/info.txt)

morrownr, 2024-01-29:
Hi @TDYK 

Interesting. Would you mind reporting this as a new issue. Using an old issue will cause me to lose track of things.

More info about your distro would be helpful.

@morrownr

closed 2024-04-02 by morrownr