morrownr/8821au-20210708 #119: Driver not working in secure boot Ubuntu 23.10

karthikvijay5227, 2024-01-11:
I have successfully installed the driver from the steps in Ubuntu 23.10. It works fine when secure boot is off, but when secure boot is on, my adapter doesn't gets recognized even though it should work according to the guide.Please give a solution. The one showing in iwconfig is my internal adapter.

[image]

morrownr, 2024-01-11:
Hi @karthikvijay5227 

Was secure boot turned on when you installed Ubuntu 23.10?

After the installation of Ubuntu 23.10, did you use the MOK managerment screen to register the keys?

Those are the two main problems I see with Ubuntu if this driver is not allowed to work due to the secure boot. Normally, if a user is installing this driver on Ubuntu, and is using `install-driver.sh` things are automatic. It just works.

I do have information in FAQ.md that you should read.

@morrownr

karthikvijay5227, 2024-01-11:
I haven't turned off secure boot at any time even during installation.There hasn't been any MOK screen during ubuntu installation.The wifi driver loads when secure boot off but not when it's on.

morrownr, 2024-01-12:
@karthikvijay5227 

Unfortunately there is a lot of variability in how Secure Boot is supported in Linux distros. Ubuntu does the best job of any distros that I used and test. Right now I do not have time to add this issue to my to-do list. I do my best to support this and the other Realtek drivers that I support on as many distros and configurations as I can but it is a massive undertaking so I just can't handle some issues as they come up.

If you are going to be running computers with Secure Boot on, your best solution, in my opinion, is to use USB WiFi adapters that use in-kernel drivers. You can find more information by going to the Main Menu of this site and reading menu items 1 and 2:

https://github.com/morrownr/USB-WiFi

There are many usb wifi adapters that use in-kernel drivers as you can see in menu item 2. That menu item is also known as the Plug and Play list.

@morrownr

closed 2024-04-02 by morrownr