morrownr/8821au-20210708 #123: Failed installation

anton-novak, 2024-03-03:
Hi, I am trying to install the driver on Xubuntu based on Ubuntu 22.04.3 LTS following the novice path and get this error message:

```
aan@aan-OptiPlex-9020M:~/Downloads/src/8821au-20210708$ sudo ./install-driver.sh 
: ---------------------------
: install-driver.sh v20231115
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 8029116 (total system memory)
: 6.5.0-14-generic (kernel version)
: gcc-12 (version of gcc used to compile the kernel)
: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
: dkms-2.8.7
: SecureBoot disabled
: ---------------------------

Checking for previously installed drivers...
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
./dkms-make.sh...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8821au: 5.12.5.2 not found
Error! Bad return status for module build on kernel: 6.5.0-14-generic (x86_64)
Consult /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 0.75 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```
This is the recommended problem reporting command output:

```
aan@aan-OptiPlex-9020M:~/Downloads/src/8821au-20210708$ sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
6.5.0-14-generic x86_64
SecureBoot disabled
Bus 002 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2357:011f TP-Link 802.11ac WLAN Adapter 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8821au/5.12.5.2: added
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

Appreciate any help! If you can recommend a Ubuntu-based Linux kernel > 6.0 distro that is tested as working with the driver, I can switch over to that as a solution.

anton-novak, 2024-03-03:
It was a fresh installation, so I migrated to OpenSUSE Tumbleweed with the latest kernel, and everything works fine now.

closed 2024-03-03 by anton-novak

morrownr, 2024-03-04:
Okay, copy all. Enjoy.