morrownr/8821au-20210708 #124: cross-compile for android

erophey7, 2024-03-10:
Hi, I'm trying to build this driver for Android
I already tried to assemble it as an out-of-tree module, and put it when compiling it in driver/net/wireless/realtek/this driver

In 1 case the following error appears:
```
erophey7@compudahter-kubuntu:~/eblya/8821au-20210708$ make -C $HOME/eblya/ksu_kernel_xiaomi_lisa O=out  ARCH=arm64 CC=clang CLANG_TRIPLE=aarch64-linux-gnu- LLVM=1 LLVM_IAS=1 CROSS_COMPILE=$HOME/tc/clang-r450784d/bin/llvm- M=$PWD modules V=1
make: Entering directory '/home/erophey7/eblya/ksu_kernel_xiaomi_lisa'
make -C /home/erophey7/eblya/ksu_kernel_xiaomi_lisa/out -f /home/erophey7/eblya/ksu_kernel_xiaomi_lisa/Makefile modules
make[1]: Entering directory '/home/erophey7/eblya/ksu_kernel_xiaomi_lisa/out'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (                \
echo >&2;                                                       \
echo >&2 "  ERROR: Kernel configuration is invalid.";           \
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";      \
echo >&2 ;                                                      \
/bin/false)
make -f ../scripts/Makefile.build obj=/home/erophey7/eblya/8821au-20210708 \
single-build= \
need-builtin=1 need-modorder=1
{  :; } \
| awk '!x[$0]++' - > /home/erophey7/eblya/8821au-20210708/modules.order
make -f ../scripts/Makefile.modpost
  sed 's/\.ko$/.lto\.o/' /home/erophey7/eblya/8821au-20210708/modules.order | scripts/mod/modpost   -i ./Module.symvers -I /home/erophey7/eblya/8821au-20210708/Module.symvers  -o /home/erophey7/eblya/8821au-20210708/Module.symvers -E    -s -T - vmlinux
make -f ../scripts/Makefile.modfinal
make[1]: Leaving directory '/home/erophey7/eblya/ksu_kernel_xiaomi_lisa/out'
make: Leaving directory '/home/erophey7/eblya/ksu_kernel_xiaomi_lisa'
erophey7@compudahter-kubuntu:~/eblya/8821au-20210708$ 
```

In 2 case the following error appears:
files:
[drivers/net/wireless/realtek/Makefile.txt](https://github.com/morrownr/8821au-20210708/files/14551804/Makefile.txt)
[drivers/net/wireless/realtek/Kconfig.txt](https://github.com/morrownr/8821au-20210708/files/14551805/Kconfig.txt)
drivers/net/wireless/realtek/rtl8812au driver path
[.config](https://github.com/morrownr/8821au-20210708/files/14551807/config.txt)

[build.sh](https://github.com/morrownr/8821au-20210708/files/14551811/build.txt)
[make_out.txt](https://github.com/morrownr/8821au-20210708/files/14551808/make_out.txt)
[modules_install.txt](https://github.com/morrownr/8821au-20210708/files/14551809/modules_install.txt)

erophey7, 2024-03-10:
UPDATE:
kernel sources: https://github.com/likkai/ksu_kernel_xiaomi_lisa
compiler: https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86/+/refs/tags/android-13.0.0_r13/clang-r450784d/

morrownr, 2024-03-10:
Hi @erophey7 

Let me merge some changes to the scripts and configuration which may help. Once that is done, I will let know and we can figure it out from there.

@morrownr

closed 2024-03-16 by erophey7

morrownr, 2024-03-16:
The changes were merged. Did it help?