morrownr/8821au-20210708 #125: Install driver without sudo

computervirus91, 2024-03-12:
I saw your recommendations after I got the adapter when it didn't work on Linux Mint. It works fine after install now.

However, I would like to install this driver without using the sudo command.

morrownr, 2024-03-13:
Hi @computervirus91 

>  I would like to install this driver without using the sudo command.

Can't say that I have seen this request before but I guess I can see where it is coming from. As far as an answer goes... I got nothing. You can't even do a basic manual installation without sudo or root.

If we were working with DOS, sure.

computervirus91, 2024-03-13:
I just wanted to be aware what the install script is doing with the sudo command in the system.
I wanted to use this driver quickly so I posted this issue :)
I was wondering if I could install it to my home/user-specific directory and create symlinks manually with sudo.

morrownr, 2024-03-13:
I understand your position but what you are proposing won't work. Drivers (or modules as they called in Linux) are part of the kernel. home directories are not.

Let me suggest this. Go to the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Scroll down to and read menu items 1 and 2. Menu item gives you some background information and menu item 2 gives you information about adapters that use in-kernel drivers. No scripts and compiling. You plug those adapters in and they will work.

computervirus91, 2024-03-13:
I read the information in the site but unfortunately I already got a TP-Link nano wifi adapter that has Realtek chipset.

The site also mentions that this driver was not developed from scratch so who developed this driver?

This repo was recommended in many websites and also has an install and remove script which is why I chose this repo.

computervirus91, 2024-03-14:
Thank you for sharing this repo and all the information that helps to get a better insight into this domain.