morrownr/8821au-20210708 #129: No combination possible (AP + managed) on RPi 4

avanmalleghem, 2024-04-04:
Hello, 

I would like to use the Alfa AWUS036ACS to create an hotspot connection and a wifi connection on a virtual interface. So having both an AP and a managed connection working in parallel on this chipset. 

To do so, I ran the commands provided by [your README](https://github.com/morrownr/8821au-20210708?tab=readme-ov-file#installation-steps) on a freshly installed RPi (Release date : March 15th 2024 - 64 bits) on which I disabled the embedded wifi interface.

The driver seems correctly installed.
Unfortunately, I can't create a virtual interface. When I use the following command : 
```
sudo iw dev wlan0 interface add wlan1 type managed
```

I have the output `command failed: No such device (-19)`

Here is the output of `iw list` : 
```
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1962
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (16.0 dBm)
			* 2417 MHz [2] (16.0 dBm)
			* 2422 MHz [3] (16.0 dBm)
			* 2427 MHz [4] (16.0 dBm)
			* 2432 MHz [5] (16.0 dBm)
			* 2437 MHz [6] (16.0 dBm)
			* 2442 MHz [7] (16.0 dBm)
			* 2447 MHz [8] (16.0 dBm)
			* 2452 MHz [9] (16.0 dBm)
			* 2457 MHz [10] (16.0 dBm)
			* 2462 MHz [11] (16.0 dBm)
			* 2467 MHz [12] (16.0 dBm)
			* 2472 MHz [13] (16.0 dBm) (no IR)
			* 2484 MHz [14] (disabled)
	Band 2:
		Capabilities: 0x862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7
		VHT Capabilities (0x03c00122):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			short GI (80 MHz)
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: not supported
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 434 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: not supported
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 434 Mbps
		VHT extended NSS: not supported
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5180 MHz [36] (13.0 dBm)
			* 5200 MHz [40] (13.0 dBm)
			* 5220 MHz [44] (13.0 dBm)
			* 5240 MHz [48] (13.0 dBm)
			* 5260 MHz [52] (13.0 dBm) (radar detection)
			* 5280 MHz [56] (13.0 dBm) (radar detection)
			* 5300 MHz [60] (13.0 dBm) (radar detection)
			* 5320 MHz [64] (13.0 dBm) (radar detection)
			* 5500 MHz [100] (13.0 dBm) (radar detection)
			* 5520 MHz [104] (13.0 dBm) (radar detection)
			* 5540 MHz [108] (13.0 dBm) (radar detection)
			* 5560 MHz [112] (13.0 dBm) (radar detection)
			* 5580 MHz [116] (13.0 dBm) (radar detection)
			* 5600 MHz [120] (13.0 dBm) (radar detection)
			* 5620 MHz [124] (13.0 dBm) (radar detection)
			* 5640 MHz [128] (13.0 dBm) (radar detection)
			* 5660 MHz [132] (13.0 dBm) (radar detection)
			* 5680 MHz [136] (13.0 dBm) (radar detection)
			* 5700 MHz [140] (13.0 dBm) (radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (13.0 dBm)
			* 5765 MHz [153] (13.0 dBm)
			* 5785 MHz [157] (13.0 dBm)
			* 5805 MHz [161] (13.0 dBm)
			* 5825 MHz [165] (13.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * tdls_mgmt
		 * tdls_oper
		 * connect
		 * disconnect
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports SAE with AUTHENTICATE command
	Device supports scan flush.
	Device supports randomizing MAC-addr in scans.
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	Supported extended features:
```

I tried on both USB2 and USB3 and kept the default options of the driver.

I tried it on a Jetson Orin Nano and had the same result. Why I think it should work is because it works on a Jetson Nano and the driver directly included in the kernel modules (kernel 4.9 - Jetpack 4.6.1). So I guess I do something wrong ?

For your information, I tried with the other driver and had the almost same issue (not the same response from iw dev) : https://github.com/aircrack-ng/rtl8812au/issues/1151
Thanks in advance for your help.

morrownr, 2024-04-04:
Hi @avanmalleghem 

In the `docs` folder is a file called `Concurrent_Mode.md`. Read it. Concurrent Mode is sort of like Virtual Interface but is not a by the book implementation. Maybe it can work for you.

When you ran `iw list`, it gave you the following line:

interface combinations are not supported

That means that this Realtek driver does not support virtual interfaces as we know them. To get more full featured adapters and drivers, you have to go with Mediatek based adapters:

https://github.com/morrownr/USB-WiFi

That is the Main Menu of this site. It has a lot of information. Menu items 1 and 2 should be interesting to Linux users that use usb wifi adapters.

Good luck,

@morrownr 

P.S. I am working on a small team right now that is trying to make a new in-kernel, standards compliant driver for 8821au. I should be posting about it in this repo soon. We will need testers.

avanmalleghem, 2024-04-05:
Thanks for the answer ! It answered to the remaining questions I had !

Following your tutorial, I have this when running `iw list` : 
```
	valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
		   total <= 2, #channels <= 1
```

Now, I have the same issue than with the version installed on Jetson Nano (v5.8). If I create a station + AP, my smartphone is not able to connect to the AP anymore (even if the one from my teammate still can [both Android]).

morrownr, 2024-04-05:
>  If I create a station + AP, my smartphone is not able to connect to the AP anymore (even if the one from my teammate still can [both Android]).

Something to keep in mind is `#channels <= 1`. That means that both station and AP have to be on the same channel as only 1 channel is supported.

Also, there could be an issue with networking setup here. tcp/ip is designed to discourage setups that are similar to repeaters because they lead to networking problems so you will need to take a hard look at your network configuration.

closed 2024-06-14 by avanmalleghem