morrownr/8821au-20210708 #130: Error compiling on Rocky Linux 9.3

bobbyl140, 2024-04-10:
<details><summary>System Info</summary>
<p>
5.14.0-362.24.1.el9_3.0.1.x86_64 x86_64

SecureBoot disabled

Bus 002 Device 005: ID 152d:1561 JMicron Technology Corp. / JMicron USA Technology Corp. JMS561U two ports SATA 6Gb/s bridge
Bus 002 Device 004: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 002 Device 003: ID 2537:1081 Norelsys NS1081
Bus 002 Device 002: ID 2109:0822 VIA Labs, Inc. USB3.1 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0489:e0a3 Foxconn / Hon Hai
Bus 001 Device 003: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 001 Device 006: ID 291a:8388 Anker Innovations Limited. Anker USB-C Hub Device
Bus 001 Device 007: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
Bus 001 Device 005: ID 14cd:8601 Super Top 4-Port hub
Bus 001 Device 002: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no

phy#1
	Interface wlan1
		ifindex 5
		wdev 0x100000001
		addr 40:49:0f:78:b8:55
		type managed
		txpower 0.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 9c:53:22:cc:81:3a
		type managed
		txpower 20.00 dBm
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#1
country 99: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(5140 - 5360 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN
	(5460 - 5860 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN

phy#0
country 99: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 40), (N/A, 20), (N/A), PASSIVE-SCAN
	(2474 - 2494 @ 40), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5140 - 5860 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN
</p>
</details> 

I am attempting to install this on Rocky Linux 9.3 to use my TP-Link Archer T2U Plus WiFi adapter, and I am getting an error while compiling that I am not familiar enough with to understand. 

Here is the `make.log`:
[make.log](https://github.com/morrownr/8821au-20210708/files/14938668/make.log)

I'm also not entirely sure if this is the best driver to use, but I've tried a few others and none have worked so far.

morrownr, 2024-04-10:
Hi @bobbyl140 

From the README:

```
Note: Red Hat Enterprise Linux (RHEL) and distros based on RHEL are supported by Red Hat devs due to the way kernel patches are handled in Red Hat. I support knowledgable RHEL developers if they want to merge the required support and keep it current. I reserve the right to delete this support without notice if it causes any problems.

Current RHEL maintainer: none
```

My opinion is that you might be better off, if running Rocky, to get an adapter that can run with in-kernel drivers. I can help you discover the in-kernel drivers that are available to you if you are interested.

bobbyl140, 2024-04-10:
Ah, I'm sorry I missed that. If you have any WiFi adapter recommendations I'd love to have them, I'm ideally looking for something that would support monitor mode either with a built in driver or a 3rd party one, but depending on the existence of said driver, it's not a requirement. Thank you!

morrownr, 2024-04-11:
> If you have any WiFi adapter recommendations I'd love to have them

We need to first figure out Rocky's kernel can do for us.

> I'm ideally looking for something that would support monitor mode either with a built in driver...

Run the following and post the result:

$ cd /usr/lib/modules/6.6.13+bpo-amd64/kernel/drivers/net/wireless/mediatek/mt76/mt76x2
$ ls mt7*

bobbyl140, 2024-04-11:
I substituted the kernel version in that command for `$(uname -r)`, as I have a different kernel version. The output is:
```
mt76x2-common.ko.xz
mt76x2e.ko.xz
mt76x2u.ko.xz
```

morrownr, 2024-04-11:
```
mt76x2-common.ko.xz
mt76x2u.ko.xz
```

That tells us there is an in-kernel driver for the mt7612u (AC1200) chipset. That probably means the mt7610u chipset is also supported but we would have to check that.

Go to:

https://github.com/morrownr/USB-WiFi/tree/main

That is the Main Menu for this site. Menu item 1 gives some good to know info and item 2 is the Plug and Play List. In that list you will find sections for recommended adapters with mt7612u and my7610u chipset.

Questions?

bobbyl140, 2024-04-11:
Nope, no questions! Thank you so much!

closed 2024-04-11 by bobbyl140

morrownr, 2024-04-11:
For monitor mode, given the drivers you have available in your kernel, I'd recommend the following:

ALFA AWUS036ACM
ALFA AWUS036ACHM

morrownr, 2024-04-11:
I happened to think about something. You will see a section in the Plug and Play List about the mt7921au chipset (WiFi 6e) or even the mt7925 chipset (WiFi 7). It is highly unlikely that Rocky has those drivers as the mt7921au driver was added to the Linux kernel with kernel 5.19 and the mt7925 driver was added to kernel 6.7. With Rocky using kernel 5.14, the mt7612u chipset and driver are the most modern available but it works well with monitor mode and is faster than most people need so it should work for you... but don't order the more modern chipsets as Rocky won't have the drivers yet.

Cheers