morrownr/8821au-20210708 #131: Wifi scan timeout is kept as 10 seconds and cannot find my hidden SSID

gustavonobrega-gsn, 2024-04-19:
Hello all.

I have a device that connects to my backend using a wifi dongle (Archer T2U and Archer T2U Nano). I was having a lot of connection drops with sites that have both 2GHz and 5GHz with same SSID and password. I was using official TP-Link driver: https://www.tp-link.com/us/support/download/archer-t2u/

Then I found you driver and my issue on reconnection was solved (I'm using commit: 847c74b1c272466f89c77947f7a9703222f0093b). But the thing is, for one of the sites my device is, the wifi connection does not work while my device uses your driver, but it works with official one. What I found different on this site is that there are a LOT SSIDs around, and my device is trying to connect to a hidden SSID. That SSID is also doubled with 2 hidden networks (2GHz and 5GHz) with same SSID and password.

My device uses wpa_supplicant for network management. When I compare wpa logs from official driver and your driver, the only thing that I see different is the scan timeout. Official driver starts with 10 seconds on the first scan, which is not enough to find that hidden SSID, then it increases to 30 seconds, which then finds the SSID. As per your driver, 10 seconds is kept forever so scan always timeout and SSID is never found, so connection never happens:

```
Scan requested (ret=0) - scan timeout 10 seconds
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
Scan timeout - try to get results
```
I wonder if I am missing some driver configuration or if there is a way to increase that timeout.

PS: I have tried hidden SSIDs in another sites with less SSIDs around and scan + connection worked fine.

**Output requested by maintainer**
5.10.209 armv7l

Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 003: ID 2357:011e

phy#1
	Interface wlanusb0
		ifindex 6
		wdev 0x100000001
		addr b0:a7:b9:70:d0:66
		type managed
		channel 3 (2422 MHz), width: 20 MHz, center1: 2422 MHz
		txpower 16.00 dBm

global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

gustavonobrega-gsn, 2024-04-19:
Other thing I have noticed, for official driver, just after scan message, I see RTM_NEWLINK messages as:

```
wlanusb0: Setting scan request: 5.000000 sec
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
wlanusb0: Starting AP scan for specific SSID: TEST SSID
wlanusb0: Add radio work 'scan'@0x4fcee8
wlanusb0: First radio work item in the queue - schedule start immediately
wlanusb0: Starting radio work 'scan'@0x4fcee8 after 0.000124 second wait
Scan requested (ret=0) - scan timeout 30 seconds
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=17 sndbuf=2000000 outq=0 send_len=45
RTM_NEWLINK: operstate=0 ifi_flags=0x1003 ([UP])
RTM_NEWLINK, IFLA_IFNAME: Interface 'wlanusb0' added
WEXT: if_removed already cleared - ignore event
Wireless event: cmd=0x8b19 len=8
wlanusb0: Event SCAN_RESULTS (3) received
Scan results did not fit - trying larger buffer (8192 bytes)
Scan results did not fit - trying larger buffer (16384 bytes)
Scan results did not fit - trying larger buffer (32768 bytes)
Received 18477 bytes of scan results (82 BSSes)
wlanusb0: BSS: Start scan result update 3
wlanusb0: BSS: Add new id 77 SSID 'TEST SSID' freq 5500
...
```

I don't see those "Interface 'wlanusb0' added" messages on your driver:

```
wlanusb0: Setting scan request: 5.000000 sec
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
wlanusb0: Starting AP scan for specific SSID: TEST SSID
wlanusb0: Add radio work 'scan'@0x52c1f8
wlanusb0: First radio work item in the queue - schedule start immediately
wlanusb0: Starting radio work 'scan'@0x52c1f8 after 0.000557 second wait
Scan requested (ret=0) - scan timeout 10 seconds
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
CTRL-DEBUG: ctrl_sock-sendto: sock=18 sndbuf=2000000 outq=0 send_len=45
Scan timeout - try to get results
wlanusb0: Event SCAN_RESULTS (3) received
Scan results did not fit - trying larger buffer (8192 bytes)
Scan results did not fit - trying larger buffer (16384 bytes)
Scan results did not fit - trying larger buffer (32768 bytes)
Received 16865 bytes of scan results (76 BSSes)
wlanusb0: BSS: Start scan result update 3
wlanusb0: BSS: Add new id 78 SSID 'Random SSID' freq 2412
...
```

morrownr, 2024-04-19:
Hi @gustavonobrega-gsn 

The TP-Link driver is older than the driver in this repo so there will be differences:

"v5.8.2.3_36567.20200401_COEX20190509-6d6f"

The driver in this repo:

"v5.12.5.2-0-g70054197b.20210708_COEX20190509-6d6f"

I am slowed with a health problem currently. Let me suggest you grep the source for interesting things that can be explored:

$ grep -r "scan" *

Let me know what you come up with.

gustavonobrega-gsn, 2024-04-22:
Hello @morrownr 

First of all I hope you recover very soon and stay well. 
Secondly thank you so much for your fast reply and for your effort on this great driver.

I understand this repo driver has a LOT of updates and improvements, so there will be a lot of differences with TP-LINK driver.

Unfortunately I cannot share much regarding scans due to site sensitive data (SSIDs and BSSIDs), but what I can tell you is I am inspecting wpa_supplicant logs, and, as the TP-LINK, this driver during the first scan already finds the BSSID of the hidden AP: 

`wlanusb0: BSS: Add new id 4 BSSID xx:xx:xx:xx:xx:xx SSID '' freq 5500`

Since this `'' `SSID does not match with my hidden AP SSID (TEST WiFi), I see other scans happening which then the TP-LINK driver finds it (after setting scan timeout to 30 seconds):

`wlanusb0: BSS: Add new id 77 BSSID xx:xx:xx:xx:xx:xx SSID 'TEST WiFi' freq 5500`

Which does not happen with this repo driver, scans continues to go on forever (always with 10 seconds), collecting a lot of SSIDs in the process, but never finding/resolving to my hidden AP SSID.

Maybe I can provide more info you might request, but I can't provide much of the site data.

Thanks in advance.

gustavonobrega-gsn, 2024-04-23:
Seems my issue was I was configuring my wpa_supplicant to work with wext driver instead of nl80211. Now everything seems to work.

Thanks again @morrownr

morrownr, 2024-04-23:
> Now everything seems to work.

Great. I had compiled the driver in this repo and plugged in my rtl8811au based adapter last night. I also changed my RasPi4B AP over to channel 149. I then tested but could not duplicate the problem.

As a side note: I have hope that we will have a Linux Wireless Standards complaint driver in the kernel at some point this year. Over the last few months a group of volunteers have been working on the in-kernel rtw88 driver for the rtl8821/11cu driver in an effort to greatly improve it. That effort has so far resulted in 5 patches that went into kernel 6.9. While additional work needs to be done, follow on projects to add rtl8821/11au and rtl8812au support to rtw88 is underway. An early Alpha is where that project is but it is not ready for testing. 

When the new driver is ready for testing, I will post a message at the top of the README here in this repo and I will start an Issue to give instructions and provide us a place to communicate. Check in periodically if you are interested.

@morrownr