morrownr/8821au-20210708 #132: Not working in Fedora 40

sauliaka, 2024-04-24:
Hi,

I happen to have a wifi dongle with rtl8811au chipset and I can't get this driver to work in it. I'm not that experienced what it comes to coding or scripting, but running ./install-driver.sh seems to give an error related to Linux kernel headers missing in latest Fedora for starters. This is probably because of the fresh (6.x+) kernel version. I was wondering if it's possible to fix it somehow, as the adapter works 100% correctly in Linux Mint with 5.15.x kernel.

Thanks in advance, if you have time to check it out.

sunwire, 2024-04-24:
You have to install kernel-devel package

sauliaka, 2024-04-24:
> You have to install kernel-devel package

is it already installed.

mothsarecool524, 2024-04-28:
> > You have to install kernel-devel package
> 
> is it already installed.

By default, no. And installing it doesn't fix the issue.

sauliaka, 2024-04-28:
> > > You have to install kernel-devel package
> > 
> > 
> > is it already installed.
> 
> By default, no. And installing it doesn't fix the issue.

I personally installed it before running that script but still there comes error. And you're correct, installing it didn't fix the issue :)

morrownr, 2024-04-28:
Hi @sauliaka 

> This is probably because of the fresh (6.x+) kernel version. 

This driver is working on all 6.x kernels that have been released and it is working on 6.9 that has not been released yet.

> I was wondering if it's possible to fix it somehow, as the adapter works 100% correctly in Linux Mint with 5.15.x kernel.

It may be possible to fix this but this is problem in your system, not with this driver as far as I can tell. You can rerun `sudo sh install-driver.sh` and copy the screen output and post it here. If the problem is indeed missing header files, that is what needs to be fixed. The best place to ask about this is probably a fedora forum.

mothsarecool524, 2024-04-28:
> Hi @sauliaka
>
>
> This driver is working on all 6.x kernels that have been released and it is working on 6.9 that has not been released yet.
>
>
> It may be possible to fix this but this is problem in [...]

[image]

I'm having the same issue here, I installed `kernel-devel` to see if that fixed it, but it did not, in my case. Same version of Fedora, 40, on the KDE Plasma spin. Relatively fresh install as well.

sauliaka, 2024-04-28:
> > Hi @sauliaka
> > > This is probably because of the fresh (6.x+) kernel version.
> > 
> > 
> > This driver is working on all 6.x kernels that have been released and it is working on 6.9 that has not been released yet.
> > > I was wondering if it's possible to fix it somehow, as the adapter works 100% correctly in Linux Mint with 5.15.x kernel.
> > 
> > 
> > It may be possible to fix this but this is problem in your system, not with this driver as far as I can tell. You can rerun `sudo sh install-driver.sh` and copy the screen output and post it here. If the problem is indeed missing header files, that is what needs to be fixed. The best place to ask about this is probably a fedora forum.
> 
> ![Screenshot_20240428_154622.png](https://private-user-images.githubusercontent.com/124483129/326302106-7f54b626-e069-4463-8cd1-24ac42bb3519.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzNDM3MTUsIm5iZiI6MTcxNDM0MzQxNSwicGF0aCI6Ii8xMjQ0ODMxMjkvMzI2MzAyMTA2LTdmNTRiNjI2LWUwNjktNDQ2My04Y2QxLTI0YWM0MmJiMzUxOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyOFQyMjMwMTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZDgxNDA5ZjVhMWRjOTU5NWMxYzUxNTYzOTRkNWQwYTg1ZGY0NzI3ZTc1MmMzZTM4Y2NiMjc3YzgxMzI0YzhjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.oI2WLXcRXXaiCSgzamJTy1hYlVajx71X8bU5DYOpeW0)
> 
> I'm having the same issue here, I installed `kernel-devel` to see if that fixed it, but it did not, in my case. Same version of Fedora, 40, on the KDE Plasma spin. Relatively fresh install as well.

This is exactly the same problem I had.

morrownr, 2024-04-28:
You only posted the error. What I want you to so is post the entire screen output of `sudo sh install-driver.sh` and please post it as text.

mothsarecool524, 2024-04-28:
`Your kernel header files aren't properly installed.
Please consult your distro documentation or user support forums.
Once the header files are properly installed, please run "sudo ./install-driver.sh"`

This is the output of the aforementioned command.

morrownr, 2024-04-28:
Give me the output of:

$ uname -r

mothsarecool524, 2024-04-28:
`6.8.7-300.fc40.x86_64`

morrownr, 2024-04-28:
Run the following:

$ sudo sh remove-driver.sh
$ make clean
$ make
$ sudo make install
$ sudo reboot

mothsarecool524, 2024-04-28:
Seems to have worked successfully, now.

morrownr, 2024-04-29:
You are welcome but now I need your help if you have time.

If I deactivate the code to detect if the header files are installed, can I get you to test it?

I need to find out what is going on in fedora that my code is not working properly. Using `sudo sh install-driver.sh` is much better than the manual installation that I showed you because it will automatically recompile the driver for you when a new kernel flows in.

mothsarecool524, 2024-04-29:
> You are welcome but now I need your help if you have time.
>
> If I deactivate the code to detect if the header files are installed, can I get you to test it?
>
> I need to find out what is going on in [...]

Sure! :D

Though another issue just popped up for me. The networks are listed with each wifi adapter name next to them, but the network I want to connect do isn't connecting on the USB adapter. (Incorrect password, even though password is correct. Network is configured correctly too.)

morrownr, 2024-04-29:
Okay. Run the following commands in the driver directory:

$ git pull
$ make clean
$ sudo sh install-driver,sh

Make sure that you reboot. See if it works.

Then we can look for the connection problem.

mothsarecool524, 2024-04-29:
It now works just fine with no issues this time. Don't know what could've been causing them, hopefully these issues don't appear for Fedora 40 users that haven't installed this driver before.

mothsarecool524, 2024-04-29:
@sauliaka Have you tried these fixes? Should work now, just `git pull` in the `~/src/` directory that was made during initial installation, and then go from there with the latest instructions from morrownr's latest comment.

sauliaka, 2024-04-29:
I managed to install it now too, wohoo :)

morrownr, 2024-04-29:
> It now works just fine with no issues this time.

> I managed to install it now too, wohoo :)

Good for both of you but not so good for me, Now I need to find out where fedora is installing its headers files. I have a lot of routines in the `install-driver.sh` script that are designed to find and fix problems or in this case, let you know there is a problem that you need to fix. Inactivating this code is solved your problem but the code was there for a good reason... header files not being installed is a big problem. Obviously your header files are installed but they are not in the standard location, which is what I check for. 

Normally header files are installed in the following location:

/lib/modules/$(uname -r)/build

Where are your header files?

sauliaka, 2024-04-29:
> Normally header files are installed in the following location:
> 
> /lib/modules/$(uname -r)/build
> 
> Where are your header files?

I believe there were 2 different versions under /lib/modules, 6.8.5-301.fc40 and 6.8.7-300.fc40. Under 6.8.5-301.fc40 build link was broken when pointing to under /usr/src/kernels/6.8.5-301.fc40, that directory didn't exist and your installer script tried to use that version. I'm here currently kinda swapping ssd drives as I'm doing some testing with my older laptop, but I can try to confirm this for you later.

laurent-onenano, 2024-06-15:
Hi everyone,

I've followed all instructions, installed, uninstalled, checked other drivers, ... Nothing worked so far.

I'm currently under Fedora 40 with kernel 6.8.11-300.fc40.x86_64
I've unfortunately bought a TP-LINK AC600 Archer T2U Nano (USB)

make (1/2) (please note that the difference in compiler in only after the last upgrade)
[image]

make (2/2)
[image]

make install
[image]

Module is located at /lib/modules/6.8.11-300.fc40.x86_64/kernel/drivers/net/wireless/8821au.ko
[image]

inxi -aN
[image]

lsusb
[image]

lsusb -t (you can see there is no driver enabled)
[image]

And still, no Wi-Fi adapter found.
[image]

Does anyone have any idea?

sunwire, 2024-06-15:
> Does anyone have any idea?

Is module loaded? (lsmod|grep  8821au)
if not try to load manually

laurent-onenano, 2024-06-15:
Good point!

It was not loaded because of SecureBoot I guess

[image]

[image]

I've tried to use the signed install but I do not have "sign" and package "sign-file" is non-existent in my repos. However, bash suggested "obs-signd" which I downloaded.

[image]

make sign-install didn't work.
[image]

I've done the 2 steps independently.
[image]

[image]

I'm rebooting now and will check if module is loaded

laurent-onenano, 2024-06-15:
Same problem

[image]

morrownr, 2024-06-15:
@laurent-onenano 

An alternative to debugging the problems with the driver in this repo is to help us test a new driver that is intended to be upstreamed into the Linux kernel soon. You can get information in the following issue:

https://github.com/morrownr/8821au-20210708/issues/133

Read the first message carefully, uninstall this driver and then follow the instructions in the README of the rtw88 repo. We know the driver is in good shape with only a few minor issues left. Your testing would be most welcome and you may find that it works better than the the driver in this repo.

Let me know how it goes.

@morrownr

laurent-onenano, 2024-06-15:
Happy to help as well mate

morrownr, 2024-06-15:
@laurent-onenano 

Great. The invitation goes to @sunwire and the others in this thread as well.

If you have any questions about installing the new rtw88 driver, let me know. I know it is a rather manual process for now but your adapters will be Plug and Play once we get it into the kernel.

sunwire, 2024-06-16:
> Same problem

If secure boot is enabled you have to:

1. Sign the module
[example](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/9/html/managing_monitoring_and_updating_the_kernel/signing-a-kernel-and-modules-for-secure-boot_managing-monitoring-and-updating-the-kernel#signing-kernel-modules-with-the-private-key_signing-a-kernel-and-modules-for-secure-boot)
2. Enroll the key

To enroll the key use command
sudo mokutil --import /path/to/public_key

4. Reboot

After restarting, a menu will appear in which you will need to enroll the key

It may helps https://youtu.be/rlCSVyv54fc
PS1. I suggest using DKMS (It will automatically build and sign the module)
PS2. If you wish you can use my COPR repo https://copr.fedorainfracloud.org/coprs/sunwire/dkms-8821au/

laurent-onenano, 2024-06-16:
It works fine!

I had another problem and it looks like the nvidia driver (from nvidia repo) created a conflict with MOK. The screen didn't appear anymore. I uninstalled the driver and all worked fine.

Thanks guys!

morrownr, 2024-06-16:
@laurent-onenano 

That's good to hear. Leave this issue open in case someone else runs into this.

You are still welcome to test the new rtw88 driver I mentioned above. The more testing we get, the sooner we can upstream it and then you won't have to compile anything.