morrownr/8821au-20210708 #134: Alfa AWUS036ACS doesn't get recognized

lSyrexl, 2024-05-05:
Hey,
I installed the driver, and encountered no errors whatsoever, but when running sudo airmon-ng it still doesn't show my AWUS036ACS for some reason.

I am using a Raspberry Pi 4 Model B (4GB RAM) that's running Raspberry Pi (OS 32-bit).

6.6.28+rpt-rpi-v8 aarch64
-bash: mokutil: command not found
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 22d4:1503 Laview Technology Glorious Model I
Bus 001 Device 003: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
8812au/5.6.4.2_35491.20191025: added
rtl8821au/5.12.5.2: added
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr de:a6:32:ef:a7:59
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr dc:a6:32:ef:a7:59
                type managed
                channel 165 (5825 MHz), width: 20 MHz, center1: 5825 MHz
                txpower 31.00 dBm
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

morrownr, 2024-05-05:
Hi @lSyrexl 

There is a lot to unpack here... but let me try to save us both a lot of back and forth messages.

My recommendation is that you burn a clean sd with:

https://www.raspberrypi.com/software/operating-systems/
Raspberry Pi OS (64-bit)
Raspberry Pi OS with desktop
Release date: March 15th 2024

Then update and upgrade and you should be in good shape to continue,

rtl8821au/5.12.5.2: added

The line above indicates to me that the compilation blew up. The RasPi devs came up with this great idea to start using a 64 bit kernel on systems where it will run even though the installed iso is 32 bit. That was a bad idea as it causes problems with many things, including what you are trying to do.

While you are here, let me point out a couple of things that might help you:

https://github.com/morrownr/USB-WiFi

That is the Main Menu of this site. The following menu items might be of help:

1. USB WiFi adapter information for Linux
2. The Plug and Play List 
9. Monitor Mode - Help with starting and using monitor mode

The two best adapters for doing wireless security work are the Alfa ACHM and ACM. You can see sections with them included in the Plug and Play List. The Alfa AXM and ACML are also good but right now there is a bug with `active` monitor mode that needs to be fixed. Regular monitor mode is good. Exactly which is best for you depends on the details of your plan. 

Also, work is underway to add a rtl8821au driver to the kernel as part of the rtw88 drivers series. Discussion is underway in issue 133 right here in this repo. Right now clean up work is underway so as to make it clean what to test so you might want to keep an eye on that issue and test when the time is right.

Regards,

@morrownr

morrownr, 2024-05-05:
@lSyrexl 

I make some editing errors in the above message that have been corrected now so you should disregard the email and read online.

closed 2024-06-08 by morrownr