morrownr/8821au-20210708 #135: Driver installs, but toolbar wifi icon doesn't show available networks

L3g3nd4ndM7th, 2024-05-11:
The driver manager detects the wifi adapter properly and recommends a driver for me to download. I downloaded and installed it without any problems. However the toolbar network icon still isn't showing me any wifi networks. In fact the computer doesn't even seem to think the wifi adapter is activated or recognized. Here are my specs below:

System:
  Kernel: 6.5.0-28-generic x86_64 bits: 64 compiler: N/A Desktop: Xfce 4.18.1 tk: Gtk 3.24.33
    info: xfce4-panel wm: xfwm 4.18.0 vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Hewlett-Packard product: HP Compaq 8200 Elite SFF PC v: N/A
    serial: <superuser required> Chassis: type: 6 serial: <superuser required>
  Mobo: Hewlett-Packard model: 1495 serial: <superuser required> UEFI: Hewlett-Packard
    v: J01 v02.33 date: 04/12/2019
CPU:
  Info: quad core model: Intel Core i5-2400 bits: 64 type: MCP smt: <unsupported>
    arch: Sandy Bridge rev: 7 cache: L1: 256 KiB L2: 1024 KiB L3: 6 MiB
  Speed (MHz): avg: 3250 high: 3284 min/max: 1600/3400 cores: 1: 3252 2: 3194 3: 3273 4: 3284
    bogomips: 24751
  Flags: avx ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3
Graphics:
  Device-1: Intel 2nd Generation Core Processor Family Integrated Graphics vendor: Hewlett-Packard
    driver: i915 v: kernel ports: active: VGA-1 empty: DP-1,HDMI-A-1 bus-ID: 00:02.0
    chip-ID: 8086:0102 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 compositor: xfwm v: 4.18.0 driver: X:
    loaded: modesetting unloaded: fbdev,vesa gpu: i915 display-ID: :0.0 screens: 1
  Screen-1: 0 s-res: 1280x1024 s-dpi: 96 s-size: 338x270mm (13.3x10.6") s-diag: 433mm (17")
  Monitor-1: VGA-1 model: HP E190i serial: <filter> res: 1280x1024 hz: 60 dpi: 87
    size: 374x299mm (14.7x11.8") diag: 484mm (19.1") modes: max: 1280x1024 min: 720x400
  OpenGL: renderer: Mesa Intel HD Graphics 2000 (SNB GT1) v: 3.3 Mesa 23.2.1-1ubuntu3.1~22.04.2
    direct render: Yes
Audio:
  Device-1: Intel 6 Series/C200 Series Family High Definition Audio vendor: Hewlett-Packard
    driver: snd_hda_intel v: kernel bus-ID: 00:1b.0 chip-ID: 8086:1c20 class-ID: 0403
  Sound Server-1: ALSA v: k6.5.0-28-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Intel 82579LM Gigabit Network vendor: Hewlett-Packard driver: e1000e v: kernel
    port: f080 bus-ID: 00:19.0 chip-ID: 8086:1502 class-ID: 0200
  IF: eno1 state: down mac: <filter>
  Device-2: NetGear A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU] type: USB driver: N/A
    bus-ID: 2-1.3:5 chip-ID: 0846:9052 class-ID: 0000 serial: <filter>
  IF-ID-1: enxf6041bdbab7d state: unknown speed: -1 duplex: half mac: <filter>
Bluetooth:
  Device-1: MediaTek N156DL type: USB driver: rndis_host v: kernel bus-ID: 1-1.5:14
    chip-ID: 0e8d:2004 class-ID: 0a00 serial: <filter>
Drives:
  Local Storage: total: 232.89 GiB used: 15.62 GiB (6.7%)
  ID-1: /dev/sda type: USB model: USB3.0 Super Speed size: 232.89 GiB type: N/A serial: <filter>
    rev: 3202 scheme: GPT
Partition:
  ID-1: / size: 227.68 GiB used: 15.61 GiB (6.9%) fs: ext4 dev: /dev/sda2
  ID-2: /boot/efi size: 511 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/sda1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
USB:
  Hub-1: 1-0:1 info: Full speed or root hub ports: 3 rev: 2.0 speed: 480 Mb/s chip-ID: 1d6b:0002
    class-ID: 0900
  Hub-2: 1-1:2 info: Intel Integrated Rate Matching Hub ports: 8 rev: 2.0 speed: 480 Mb/s
    chip-ID: 8087:0024 class-ID: 0900
  Device-1: 1-1.5:14 info: MediaTek N156DL type: Bluetooth,CDC-Data driver: rndis_host
    interfaces: 2 rev: 2.0 speed: 480 Mb/s power: 500mA chip-ID: 0e8d:2004 class-ID: 0a00
    serial: <filter>
  Device-2: 1-1.7:4 info: China Resource Semico USB Keyboard type: Keyboard,Mouse
    driver: hid-generic,usbhid interfaces: 2 rev: 1.1 speed: 1.5 Mb/s power: 500mA
    chip-ID: 1a2c:4c5e class-ID: 0301
  Device-3: 1-1.8:5 info: [Maxxter] Optical Gaming Mouse [Xtrem] type: Mouse,Keyboard
    driver: hid-generic,usbhid interfaces: 2 rev: 1.1 speed: 1.5 Mb/s power: 100mA
    chip-ID: 18f8:0f97 class-ID: 0300
  Hub-3: 2-0:1 info: Full speed or root hub ports: 3 rev: 2.0 speed: 480 Mb/s chip-ID: 1d6b:0002
    class-ID: 0900
  Hub-4: 2-1:2 info: Intel Integrated Rate Matching Hub ports: 6 rev: 2.0 speed: 480 Mb/s
    chip-ID: 8087:0024 class-ID: 0900
  Device-1: 2-1.2:4 info: JMicron / USA JMS567 SATA 6Gb/s bridge type: Mass Storage driver: uas
    interfaces: 1 rev: 2.1 speed: 480 Mb/s power: 500mA chip-ID: 152d:0562 class-ID: 0806
    serial: <filter>
  Device-2: 2-1.3:5 info: NetGear A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU] type: WiFi
    driver: N/A interfaces: 1 rev: 2.1 speed: 480 Mb/s power: 500mA chip-ID: 0846:9052
    class-ID: 0000 serial: <filter>
Sensors:
  System Temperatures: cpu: 44.0 C mobo: N/A
  Fan Speeds (RPM): N/A
Repos:
  Packages: apt: 2358
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/vscode.list
    1: deb [arch=amd64,arm64,armhf] http: //packages.microsoft.com/repos/code stable main
Info:
  Processes: 260 Uptime: 7h 45m wakeups: 0 Memory: 7.64 GiB used: 4.09 GiB (53.5%) Init: systemd
  v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12 Client: Unknown python3.10 client
  inxi: 3.3.13

onestan, 2024-05-15:
I've had the same experience using the latest Fedora 40 and OpenSUSE Tubmleweed releases with GNOME

jap-gln, 2024-08-06:
I resolved this issue by connecting my wifi adapter to port USB 3.0.

Switch of the machine
Go to Settings 
Click on USB
Enable USB Controller
Select USB 3.0
Under USB Device filters. Add USB device filter. Select Realtek  802.11ac WLAN adapter.
Press Ok.

morrownr, 2024-08-06:
> The driver manager detects the wifi adapter properly and recommends a driver for me to download. 

> Switch of the machine

Is this report about the driver in this repo? It is hard to help when I can't even make out what is going on.

FYI: There is a really good in-kernel driver for the rtl8821/11au chip going into the kernel very soon. Would you like to test it?