morrownr/8821au-20210708 #137: Driver not working after update

Stachelbeere1248, 2024-05-17:
Yesterday I updated my zypper packages on OpenSUSE Tumbleweed, resulting in updating this driver (rtl8821au) as well as the kernel, however it appears to not find any wireless networks. What information can I provide to get some help, and could this be related to this new package "rtw88"?

_Originally posted by @Stachelbeere1248 in https://github.com/morrownr/8821au-20210708/issues/133#issuecomment-2117075075_
            
            
I updated the kernel from 6.8.7-1-default to 6.8.9-1-default, along with the package from my package manager. What additional information can I provide? [here](https://github.com/morrownr/8821au-20210708/files/15355266/temp.txt) is the journalctl if that helps.

belzerus, 2024-05-17:
The log indicated that you cannot find the firmware.

```
hiddev100,hidraw8: USB HID v1.11 Device [ROCCAT ROCCAT Kone Pro] on usb-0000:00:14.0-8/input3
May 17 11:20:28 localhost kernel: rtw_8821au 1-4:1.0: Direct firmware load for rtw88/rtw8821a_fw.bin failed with error -2
May 17 11:20:28 localhost kernel: rtw_8821au 1-4:1.0: failed to request firmware
May 17 11:20:28 localhost kernel: 00000000: 29 81 00 7c 01 00 01 00 4c 00 04 00 10 00 00 00  )..|....L.......
May 17 11:20:28 localhost kernel: 00000010: 26 26 26 26 26 26 2f 2f 2f 2f 2f ff ff
```

Can you double-check that the firmware exists on the platform and what permission and checksum it has? Send back output of:

```
ls -la /lib/firmware/rtw88/rtw8821a_fw.bin
```

and

```
md5sum /lib/firmware/rtw88/rtw8821a_fw.bin
```

lwfinger, 2024-05-17:
If those commands result in the file not being found, you need to run the following command from the rtw88 directory:

sudo cp rtw8821a_fw.bin /lib/firmware/rtw88/.

Stachelbeere1248, 2024-05-19:
The file is actually not found, however I am not entirely sure if an rtw88 directory with this file even exists. Would using the one from the rtw88 repo just work?

morrownr, 2024-05-19:
I finally had time to stop by and read this thread. Maybe I am lost but it sure looks to me like @Stachelbeere1248 is reporting a problem with the vendor driver in this repo and not with rtw88. We need to figure that out before offering help.

@Stachelbeere1248 

Where did you get the driver that you are using?

Stachelbeere1248, 2024-05-19:
As I said in the first message of this thread I installed everything through my package manager zypper on OpenSUSE tumbleweed, specifically this repo https://download.opensuse.org/repositories/home:/Sauerland:/hardware/openSUSE_Tumbleweed/

lwfinger, 2024-05-20:
@morrownr - That driver is from rtw88 including Bitterblue's most recent changes. Sauerland packages the kernel modules for rtw89 and rtw88 for the openSUSE distributions.

@Stachelbeere1248 - The firmware has been submitted to linux-firmware, but not yet merged. You should clone https://github.com/lwfinger/rtw88.git, cd to the directory, and run the cp command from my previous post.

morrownr, 2024-05-20:
@lwfinger 

> That driver is from rtw88 including Bitterblue's most recent changes. Sauerland packages the kernel modules for rtw89 and rtw88 for the openSUSE distributions.

Thanks for the explanation. There are several distros that package the vendor driver in this repo but I was unaware that any distros were packaging your rtw88. Sorry for stepping in. I will depart now.