morrownr/8821au-20210708 #138: Two interfaces after installing + tainting kernel

Asaf-E, 2024-05-28:
Hi,

After installing the driver I have two interfaces:
nmcli dev
wlan0          wifi      disconnected  --
p2p-dev-wlan0  wifi-p2p  disconnected  --

 iw dev
phy#0
        Interface wlan0
                ifindex 5
                wdev 0x1
                addr 08:be:ac:3b:eb:eb
                type managed
                txpower -100.00 dBm
                
Is there a method to disable the second interface from the device parameters?

Additionaly, I have the error:
dmesg | grep 8821au
[    9.428902] 8821au: loading out-of-tree module taints kernel.
[    9.428974] 8821au: module verification failed: signature and/or required key missing - tainting kernel
[   11.439217] usbcore: registered new interface driver rtl8821au

I don't have "secure boot", yet this still shows, I checked and the module is signed:
modinfo 8821au | grep '^sig'
sig_id:         PKCS#7
signer:         javelin Secure Boot Module Signature key
sig_key:        1C:79:D5:54:B7:92:66:ED:DA:97:6E:7D:42:77:B2:CF:18:08:18:FF
sig_hashalgo:   sha512
signature:      34:B0:A5:8E:A7:59:EE:71:BE:2A:98:8D:6D:1A:72:5E:DD:27:17:2F:

I'm using the kernel:
uname -r
6.5.0-35-generic

I used the automatic installation.

I have been able to connect to wifi using the wlan0 interface, but I'm worried about these issues effect.

this is the dkms:
dkms status
r8168/8.049.02, 5.19.0-46-generic, x86_64: installed
r8168/8.049.02, 6.5.0-35-generic, x86_64: installed
rtl8821au/5.12.5.2, 6.5.0-35-generic, x86_64: installed

morrownr, 2024-05-28:
Thanks for moving.

> [ 9.428902] 8821au: loading out-of-tree module taints kernel.

I can't do anything about this. This is the Linux kernel maintainers telling you to stop using Realtek out-of-kernel drivers because they are not standards compliant. The driver I asked you to test is standards compliant and will be going into the kernel.

> iw dev
phy#0
Interface wlan0
ifindex 5
wdev 0x1
addr 08:be:ac:3b:eb:eb
type managed
txpower -100.00 dBm

I only see one interface.

What hardware are you using? Is this a RasPi or what?

> I have been able to connect to wifi using the wlan0 interface, but I'm worried about these issues effect.

Don't worry. There is nothing we can do as long as you are on this driver. If it works, it works.

Do you have the links to the new standards compliant driver that I posted?

Asaf-E, 2024-05-28:
I'm using a MiniPC X100-N3350

I haven't yet installed the mac80211 driver

morrownr, 2024-05-28:
> MiniPC X100-N3350

Cool. Been thinking about getting a MiniPC myself. So this is a x86_64 based system.

Helping someone when you are not there can be a challenge but I do the best that I can.

It sounds to me like this driver is working. If and when you get ready to try the new driver, let me know.

closed 2024-06-08 by morrownr