morrownr/8821au-20210708 #139: Rtl8811au driver instantly CRASHES VM

bigjoncoop, 2024-06-02:
###  _**REQUESTED INFO:**_
**WiFi Adapter:** Alpha AWUS036ACS
**Chipset:** RTL8811AU
**OS:** KALI 2024.1 - Official Pre-Made Kali VirtualBox VM (Latest Update / Upgrade as of 6/2/2024)
**Hardware/Software:** HP EliteBook 745 G5  Ryzen 7 - Windows 11 - Virtual Box 7.0.18 - Extention Pack 7.0.18 

**Requested Terminal Output:**
(only had a few seconds to copy this after inserting the wifi adapter, before the VM crashes...) 
```6.6.15-amd64 x86_64
EFI variables are not supported on this system
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 003: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]
realtek-rtl8814au/5.8.5.1~git20231211.866a910, 6.6.15-amd64, x86_64: installed
realtek-rtl88xxau/5.6.4.2~git20231117.a3e0c0b, 6.6.15-amd64, x86_64: installed
rtl8821au/5.12.5.2, 6.6.15-amd64, x86_64: installed
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

### _**ISSUE:**_

Kali runs perfectly fine in the Virtualbox VM with out having the RTL8811AU wifi adapter enabled. Also if I connect a different USB wifi adapter, for instance one that uses the RTL8812BU chipset, the VM doesn't crash and everything works as it should. 

Before installing the RTL8811AU drivers by following your guide, the Wi-Fi dongle would not work. It would show up in the list if I did `lsusb` but that's it. Since after installing your drivers the Wi-Fi dongle will now show up as a usable device but the VM will crash within 10 seconds of an enabling/passing it thru to the VM. The VM will just 100% freeze and I need to force close the window / end the windows process.

If you need me to provide any other info or specific LOG files just let me know...

*I have also installed the RTL8811AU driver by following your guide on a baremetal machine running KALI (which did cause some display issues though), and the driver did work allowing me to use this USB wifi dongle.  So I know the your driver and this wifi adapter do work. 

THANKS!!!

morrownr, 2024-06-03:
@bigjoncoop 

My plan for this out-of-kernel driver is to continue API related changes until the in-progress in-kernel driver is upstreamed. Soon after that, I'll look at retiring this driver. Let me invite you to test the new in kernel driver. The location where we are coordinated the effort and taking bug reports is right here in this repo:

https://github.com/morrownr/8821au-20210708/issues/133

The first message gives details and the location of the repo that has the code to test. It is a manual installation with no scripts unlike this driver but it is not hard to install. Read the README. Don't forget to uninstall this driver first via:

$ sudo sh remove-driver.sh

Hopefully I see in the above shown issue soon,

@morrownr

morrownr, 2024-06-08:
@bigjoncoop 

Have you been able to test the new driver? A big drop of patches took place yesterday. The driver is in really good shape. We can use all of the testers/reporters we can get. It would be good to know if the new driver fixes your problem.

Thanks