morrownr/8821au-20210708 #141: Build failure alma 9.4

zenonp, 2024-06-16:
kernel, kernel-devel and kernel-headers 5.14.0-427.20.1, dkms 3.0.13-1 gcc and kernel gcc matching 11.4.1 20231218. Build log attached. 

[make.log](https://github.com/user-attachments/files/15858143/make.log)
[screen.log](https://github.com/user-attachments/files/15858146/screen.log)

morrownr, 2024-06-16:
Hi @zenonp 

From the README:

Note: Red Hat Enterprise Linux (RHEL) and distros based on RHEL are supported by Red Hat devs due to the way kernel patches are handled in Red Hat. I support knowledgable RHEL developers if they want to merge the required support and keep it current. I reserve the right to delete this support without notice if it causes any problems.

Current RHEL maintainer: none

zenonp, 2024-06-16:
Hi

The problems start in osdep_service.c at this point:

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 17, 0))
        kthread_complete_and_exit(comp, 0);
#else
        complete_and_exit(comp, 0);
#endif

And we know that RH kernels include quite a lot of backports, so what RH calls "5.14.0" is likely a mishmash of 5.14 and various later kernels. I tried #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 14, 0)) and osdep_service.o was built successfully, only for the build to break a bit later at 

os_intfs.c:2142:9: error: too many arguments to function ‘netif_napi_add’
 2142 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);

Now, I can keep testing until I weed out all the errors, but I'm not a developer and I don't know how to patch your code with the right "if rhel then this, else that"-conditionals. Would you be willing to do that part?

morrownr, 2024-06-16:
@zenonp 

I can offer suggestions if you want usb wireless.

1. The 88x2bu driver here at this site has a RHEL maintainer so you could use it as an example if you want to fork the 8821au driver:

https://github.com/morrownr/88x2bu-20210702

2. We could take a look at the files that Alma includes to see what usb wifi adapters are already supported or can be easily supported with some installed files. This is likely the easiest solution and I would like to know myself so I can help you dig for this info.

3. I am working with a team to upstream a standards compliant driver for the 8821au. It is almost ready but may require some mods like you would see in the 88x2bu driver... that I would be willing to help with.

Think about it and let me know what option or options you want to explore.

zenonp, 2024-06-21:
> I'll start the weeding and then see where I get.

I've been a bit overwhelmed by other stuff, so it took time. But it's done. With the patch below the module compiles and works with a 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]. In fact I'm posting this through the T2U running on the modified 8821au-20210708 that I just compiled. 

So now we know what kernel version(s) a Redhat 5.14.0-427.20.1 really is. From there on, it would take someone with more knowledge of RH kernels and more dedication than myself, to create and maintain an el9-compatible version of the code. 

> We could take a look at the files that Alma includes to see what usb wifi adapters are already supported 

I needed a driver for the T2U. I bought it by mistake, thinking that it had in-kernel support, and ended up here instead. It was only €15, I could have put in the pile of "things that might be useful some other day", but making it work was not so far away and it was definitely more satisfactory. 

====
```
diff -Nur 8821au-20210708/os_dep/linux/ioctl_cfg80211.c 8821au-20210708-el9fixed/os_dep/linux/ioctl_cfg80211.c
--- 8821au-20210708/os_dep/linux/ioctl_cfg80211.c       2024-06-21 23:12:15.997470004 +0300
+++ 8821au-20210708-el9fixed/os_dep/linux/ioctl_cfg80211.c      2024-06-21 23:00:58.569013615 +0300
@@ -471,7 +471,8 @@
                cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false);
 #endif
 #else
-               cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, false);
+               cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false, 0);
+//             cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, false);
 #endif
 #else
                cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0);
@@ -490,7 +491,8 @@
        cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
 #endif
 #else
-       cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
+//     cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
+       cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0, 0);
 #endif
 
 #else
@@ -1162,7 +1164,8 @@
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 0, 0))
                roam_info.links[0].bssid = cur_network->network.MacAddress;
 #else
-               roam_info.bssid = cur_network->network.MacAddress;
+//             roam_info.bssid = cur_network->network.MacAddress;
+               roam_info.links[0].bssid = cur_network->network.MacAddress;
 #endif
                roam_info.req_ie = pmlmepriv->assoc_req + sizeof(struct rtw_ieee80211_hdr_3addr) + 2;
                roam_info.req_ie_len = pmlmepriv->assoc_req_len - sizeof(struct rtw_ieee80211_hdr_3addr) - 2;
@@ -1923,9 +1926,9 @@
 }
 
 static int cfg80211_rtw_add_key(struct wiphy *wiphy, struct net_device *ndev
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 1, 0))
+//#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 1, 0))
        , int link_id
-#endif
+//#endif
        , u8 key_index
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 37)) || defined(COMPAT_KERNEL_RELEASE)
        , bool pairwise
@@ -2088,9 +2091,9 @@
 }
 
 static int cfg80211_rtw_get_key(struct wiphy *wiphy, struct net_device *ndev
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 1, 0))
+//#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 1, 0))
        , int link_id
-#endif
+//#endif
        , u8 keyid
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 37)) || defined(COMPAT_KERNEL_RELEASE)
        , bool pairwise
@@ -2279,9 +2282,9 @@
[... 137 lines cut ...]
+//     wiphy->regulatory_flags |= REGULATORY_IGNORE_STALE_KICKOFF;
+//#endif
 
        return 0;
 }
diff -Nur 8821au-20210708/os_dep/osdep_service.c 8821au-20210708-el9fixed/os_dep/osdep_service.c
--- 8821au-20210708/os_dep/osdep_service.c      2024-06-21 23:12:16.001470038 +0300
+++ 8821au-20210708-el9fixed/os_dep/osdep_service.c     2024-06-16 20:17:35.782152392 +0300
@@ -1309,11 +1309,11 @@
 inline void thread_exit(_completion *comp)
 {
 #ifdef PLATFORM_LINUX
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 17, 0))
+//#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 17, 0))
        kthread_complete_and_exit(comp, 0);
-#else
-       complete_and_exit(comp, 0);
-#endif
+//#else
+//     complete_and_exit(comp, 0);
+//#endif
 #endif
 
 #ifdef PLATFORM_FREEBSD
```

zenonp, 2024-06-22:
The problem with RH kernels is that "(LINUX_VERSION_CODE >= KERNEL_VERSION(x, y, z))" means absolutely nothing and can never be trusted. So it occurs to me that the way to get around this, is to test directly for what some particular kernel expects. Say, with the very first code change above, test whether adapter-> expects four of five arguments and then do accordingly. I don't know whether this could be done, yet less how, but it would be universal and cover any kernel of any distribution, patched or not.

morrownr, 2024-06-22:
Hey, I deal with api changes regularly so I understand the challenges... but then Red Hat comes along and mucks things up. If I were to try to support RHEL, it would take many additional hours of my time and I already do not have enough time to do what I want to do.

My suggestion is that we investigate what drivers for wifi chips are already being made available and look in that direction.

Run the following commands and then copy and paste the results here:

$ cd /usr/lib/modules/$(uname -r)/kernel/drivers/net/wireless/mediatek/mt76
$ ls -l

zenonp, 2024-06-23:
There's a mix of files and directories there, so therefore 
```
# pwd
/usr/lib/modules/5.14.0-427.20.1.el9_4.x86_64/kernel/drivers/net/wireless/mediatek/mt76

# find ./ -type f -exec ls -l {} \;
-rw-r--r--. 1 root root 31176 Jun 10 15:34 ./mt76x2/mt76x2u.ko.xz
-rw-r--r--. 1 root root 25600 Jun 10 15:34 ./mt76x2/mt76x2-common.ko.xz
-rw-r--r--. 1 root root 31916 Jun 10 15:34 ./mt76x2/mt76x2e.ko.xz
-rw-r--r--. 1 root root 33712 Jun 10 15:34 ./mt76x0/mt76x0-common.ko.xz
-rw-r--r--. 1 root root 26440 Jun 10 15:34 ./mt76x0/mt76x0e.ko.xz
-rw-r--r--. 1 root root 26080 Jun 10 15:34 ./mt76x0/mt76x0u.ko.xz
-rw-r--r--. 1 root root 23496 Jun 10 15:34 ./mt76x02-usb.ko.xz
-rw-r--r--. 1 root root 44048 Jun 10 15:34 ./mt792x-lib.ko.xz
-rw-r--r--. 1 root root 56808 Jun 10 15:34 ./mt76x02-lib.ko.xz
-rw-r--r--. 1 root root 67844 Jun 10 15:34 ./mt76.ko.xz
-rw-r--r--. 1 root root 30836 Jun 10 15:34 ./mt76-usb.ko.xz
-rw-r--r--. 1 root root 59532 Jun 10 15:34 ./mt7921/mt7921-common.ko.xz
-rw-r--r--. 1 root root 29320 Jun 10 15:34 ./mt7921/mt7921e.ko.xz
-rw-r--r--. 1 root root 75332 Jun 10 15:34 ./mt7925/mt7925-common.ko.xz
-rw-r--r--. 1 root root 29336 Jun 10 15:34 ./mt7925/mt7925e.ko.xz
-rw-r--r--. 1 root root 56452 Jun 10 15:34 ./mt76-connac-lib.ko.xz

```

zenonp, 2024-06-23:
> Red Hat comes along and mucks things up

Yes, but there must be some way to easily figure out what a kernel expects. For example my last edit above, kthread_complete_and_exit or complete_and_exit, could be picked up directly from System.map without knowing the kernel version. And it should be possible to find whether int link_id should be used or not by looking in cfg80211.h. Things like that. 

The benefit of this method is not so much that it would support RH kernels (RH should do that themselves), but that it is version-independent and 100% fail-safe. 

> it would take many additional hours of my time

I understand that and I'm not asking you to do anything. I'm just putting forward ideas that could be picked up by you, by someone else, or even by myself, now or at some other point in time. 

BTW, looking at install_driver.sh I just realised that I don't need to learn C to do this; I can probably do it in bash with a few grep and sed commands from install_driver.sh itself.

morrownr, 2024-06-23:
> /mt76x2/mt76x2u.ko.xz

There is a driver for the mt7612u AC1200 chip.

> /mt76x0/mt76x0u.ko.xz

There is a driver for the mt7610u AC600 chip.

> /mt7921/mt7921e.ko.xz

That is the pcie driver for the mt7921au AXE3000 chip. There is also a usb version of the driver. It is not clear why it is not being compiled.

So, you do have options. The Main Menu for this site is: 

https://github.com/morrownr/USB-WiFi

Menu item 2 is the Plug and Play List. It has sections on the mt7612u and mt7610u based adapters.

There is also the option to investigate an effort to upstream a new 8821/11au driver:

https://github.com/morrownr/8821au-20210708/issues/133

The first message in that thread gives information on the project and points you to the repo where the code is being worked. It is a manual installation but that is just for testing. Once the driver is in the kernel, it will be plug and play. You may run into the same issues with this driver that you do with the driver in this repo that you are trying to compile. I can understand why Red Hat would have thought what they are doing to the kernel was a good idea at one time but it is no longer a good idea and just causes problems.

zenonp, 2024-06-23:
> There is a driver for the mt7612u AC1200 chip.

Thank you for the suggestions, I appreciate your effort. 

On the downside, I am currently in Greece where none of these adapters seem to be on sale (or they're very hard to find because of a widespread indifference in this country towards giving exact technical specifications). I found them and I can get them from elsewhere in the EU, but then shipping gets pretty steep. 

On the upside, my Archer T2U now works just fine. So my focus has shifted from getting the adapter to work, to getting this driver to compile and work no matter what. 

> I can understand why Red Hat would have thought what they are doing to the kernel was a good idea at one time but it is no longer a good idea and just causes problems. 

I can see their point in phasing in changes gradually and conservatively, but after all these years they should at least have figured an easy way, even a fully-fledged ABI, to tell the user what code he is running.

darkranger-red, 2024-12-05:
I am not sure if I should open a new issue or remain on this one. But anyway, I did some revision to make the current codebase compilable on RHEL 9.5 kernels (tested on kernel-5.14.0-503.14.1.el9_5.x86_64):
https://github.com/darkranger-red/8821au-20210708/commit/96c2a8f8d4fb6f68fd401516fc6bf3b30e58d3e7

I referred to the code from the project `88x2bu-20210702` to see how the RHEL compatibility being implemented, and took a little different approach to identify the RHEL releases. For instance, just like to execute the command `cat /usr/src/kernels/$(uname -r)/include/generated/uapi/linux/version.h` to obtain the following information on the recent CentOS Stream 9 distribution:
```
#define LINUX_VERSION_CODE 331264
#define KERNEL_VERSION(a,b,c) (((a) << 16) + ((b) << 8) + ((c) > 255 ? 255 : (c)))
#define LINUX_VERSION_MAJOR 5
#define LINUX_VERSION_PATCHLEVEL 14
#define LINUX_VERSION_SUBLEVEL 0
#define RHEL_MAJOR 9
#define RHEL_MINOR 6
#define RHEL_RELEASE_VERSION(a,b) (((a) << 8) + (b))
#define RHEL_RELEASE_CODE 2310
#define RHEL_RELEASE "533"
```

But I am afraid that to test and revise the code accordingly on both RHEL 8 and 9 and also every minor release is simply out of my capability. Since I only have one USB WiFi adapter(TP-Link Archer T2U Plus) and one specific RHEL 9.x machine needs this driver to work for now. It seems to be more practical to just make the code compilable on the latest RHEL 9 releases.

morrownr, 2024-12-05:
Hi @darkranger-red 

> I am not sure if I should open a new issue...

It would be best if you open a new issue as we may not be able to find this in this location.

> I did some revision to make the current codebase compilable on RHEL 9.5 kernels

I see. Nice work.

> I referred to the code from the project 88x2bu-20210702 to see how the RHEL compatibility being implemented

That was a good idea. The maintainer of the RHEL support in the 88x2bu driver does a good job.

> It seems to be more practical to just make the code compilable on the latest RHEL 9 releases.

That makes sense.

> and took a little different approach to identify the RHEL releases.

I see that. I can appreciate your method but I would prefer a standard method be used on the drivers here for consistency...unless you see a potential problem. If you want to rework your patch to use the method used in 88x2bu and make a PR, I will be glad to merge it.

darkranger-red, 2024-12-06:
Hello @morrownr 

At first I was intent to do exactly the same method as the Makefile from `88x2bu-20210702`. But it seems unnecessary tricky to me to depend on the content of `/etc/os-release`. Since I am not a kernel module developer, maybe there is a better reason behind this design and I just don't realize yet. So I think it might be good to raise a discussion here.

The main concern is that I think we should depend on the information directly from the kernel source tree as much as possible during the kernel module compilation, rather than something under `/etc/` which the user or program could have some reason to touch it.

Let's say, here is one of the unlikely but still not impossible scenarios to make `/etc/os-release` become a little imprecise:
https://access.redhat.com/solutions/7079883

To depend on `/etc/os-release` is indeed works, and could work permanently. But if we depend on `linux/version.h` instead, I think the concept is more clear and more straightforward. If `linux/version.h` does exist and there is `RHEL_MAJOR` inside - it's definitely a RHEL kernel. No longer need to look for the string "fedora" and get the RHEL kernel release number from somewhere else. Because `linux/version.h` already has all of the information (`RHEL_MAJOR`, `RHEL_MINOR` and `RHEL_RELEASE`) we need to look for. So I just grabbed them all into the Makefile, to provide more flexibility for the future maintenance.