morrownr/8821au-20210708 #143: Error 2 installing on Kali

BPCDoug, 2024-06-28:
Here is what I am getting installing on latest 2024.2 Kali on VMware VM 

Thanks in advance. I am not a pro on Linux, it maybe user error, but have tried all the basics of updates etc...

starting installation.
Installing 8821au.conf to /etc/modprobe.d
The non-dkms installation routines are in use.
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.8.11-amd64/build M=/home/kali/src/8821au-20210708  modules
make[1]: *** /lib/modules/6.8.11-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2498: modules] Error 2
An error occurred:  2
Please report this error.

morrownr, 2024-07-05:
Hi @BPCDoug 

Sorry for the delay. I have been very busy. Do you still need help?

Vspoke91, 2024-07-13:
I'm getting this same error, any fix?

morrownr, 2024-07-13:
Hi @Vspoke91 

Let me suggest an alternative. Uninstall this driver:

$ sudo sh remove-driver.sh

Then go into the following issue and read the first message:

https://github.com/morrownr/8821au-20210708/issues/133

We have been working on a new driver that can be upstreaming into Linux mainline. Try it. The installation is different than this repo so read the installation instructions.

Good luck

Darthscion55, 2024-07-19:
> Hi @Vspoke91
>
> Let me suggest an alternative. Uninstall this driver:
>
> $ sudo sh remove-driver.sh
>
> Then go into the following issue and read the first message:
>
> #133
>
> We have been working on a new [...]

AAA can you maybe add something to your main page to let people know this? I got this working a long time ago for mint, (can't remember which repo) and now need or for kali. I have been struggling for days. Thank you guys for all your hard work!

morrownr, 2024-07-19:
@Darthscion55 

> can you maybe add something to your main page to let people know this?

The very first line of the README.md already does this.