morrownr/8821au-20210708 #144: Installation successful but no wifi

FoolHen, 2024-06-30:
Hi, I have installed the driver without issues on Debian 11 Bullseye (5.10.0-30-amd64 x86_64). I have a usb stick that needs RTL8811A/RTL8821A USB drivers. However, after installing it wifi is sill not enabled. I have tried different ports but didn't make a difference.

Not sure if this issue is valid, as the driver installed correctly. The device is from an unknown brand (https://approx.es/producto/appusb600da-adaptador-usb-wifi-600mb-antena/) and they have linux drivers (software tab) but they are outdated, so I was not able to install them. However, it includes some documentation where it states that need RTL8811A/RTL8821A USB drivers.

Any recommendation on how to check whats wrong?

running lsusb -v gives me:

<details>

<summary>lsusb -v</summary>

### You can add a header

Bus 001 Device 005: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xc811 
  bcdDevice            2.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

</details>

```

morrownr, 2024-06-30:
Hi @FoolHen 

> it includes some documentation where it states that need RTL8811A/RTL8821A USB drivers.

That documentation is wrong. The VID/PID ID 0bda:c811 indicates that your adapter uses the rtl8821/11cu chip so you need to clean out/remove any incorrect drivers and go to:

https://github.com/morrownr/8821cu-20210916

FoolHen, 2024-07-10:
Thank you, that one seemed to partially work. I'm able to now see the wireless networks but unfortunately it can't connect to any network, their settings seem fine but it keeps asking for password and retrying. It's probably the adapter's fault, right?