morrownr/8821au-20210708 #146: gcc --version gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0

Khnaz35, 2024-07-07:
Hi @morrownr
[https://github.com/morrownr/88x2bu-20210702/issues/224#issuecomment-2211433556](url)

As suggested on other repo i did make test and follow the Readme.md but seems there will be issues because gcs version is mismatched.

```
$ sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
6.8.0-36-generic x86_64
EFI variables are not supported on this system
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 003: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 004: ID 2357:011f TP-Link 802.11ac WLAN Adapter 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 222a:0001 ILI Technology Corp. Multi-Touch Screen
Bus 003 Device 003: ID 0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)
Bus 003 Device 004: ID 0000:0538   USB OPTICAL MOUSE
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```
```
$ mkdir -p ~/src
@khnaz35:~$ cd ~/src
@khnaz35:~/src$ git clone https://github.com/morrownr/8821au-20210708.git
Cloning into '8821au-20210708'...
remote: Enumerating objects: 1659, done.
remote: Counting objects: 100% (222/222), done.
remote: Compressing objects: 100% (75/75), done.
remote: Total 1659 (delta 162), reused 190 (delta 147), pack-reused 1437
Receiving objects: 100% (1659/1659), 3.44 MiB | 10.49 MiB/s, done.
Resolving deltas: 100% (862/862), done.
@khnaz35:~/src$ cd ~/src/8821au-20210708
```
```
@khnaz35:~/src/8821au-20210708$ cat /proc/version
Linux version 6.8.0-36-generic (buildd@lcy02-amd64-077) (x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #36-Ubuntu SMP PREEMPT_DYNAMIC Mon Jun 10 10:49:14 UTC 2024
@khnaz35:~/src/8821au-20210708$ gcc --version
gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Khnaz35, 2024-07-07:
Just update i went ahead and still tested
```
sudo ./install-driver.sh
```
There were few recommendations and even option to edit the file while ignoring all by pressing **N** seems driver is installed.

```
$ dkms status
rtl8821au/5.12.5.2, 6.8.0-36-generic, x86_64: installed
```

morrownr, 2024-07-07:
@Khnaz35 

It looks to me that everything worked fine. If it were me, I would have rebooted after the installation finished. Is it working?

Khnaz35, 2024-07-08:
@morrownr 
I have rebooted the system and connected to my network. Everything seems to be working fine now. If any issues arise, I'll let you know. Thanks for your help!

closed 2024-07-08 by Khnaz35