morrownr/8821au-20210708 #151: Driver crashes PC on Arch Linux 6.10.8-arch1-1

c0ncatenate, 2024-09-07:
Hello,

I am currently on **Arch Linux** kernel **6.10.8-arch1-1** and I have an **Alfa AWUS036ACS AC600 Wireless Adapter**. [DeviWiki](https://deviwiki.com/wiki/Iconnect_AWUS036ACS) suggests that my adapter uses the RTL8812AU driver. I've installed the driver using the `install.sh` script provided and everything installs as intended.

The device seems to recognize the driver and the LED lights up. I can see it in my WiFi devices list. However, when I try to use an app like Wireshark, it crashes my entire PC. When I try using tools such as `sudo airodump-ng wlan1`, it is unable to find any APs around me. I believe the driver is not updated to the latest kernel on Arch and it's breaking something in the background. I have also tried the driver from [Aircrack-NG](https://github.com/aircrack-ng/rtl8812au) but it doesn't work for me either.
I have also tried this with arch 6.6.49-1 and it's still the same.
If there's an updated driver/kernel that I can use or a workaround to get it working, please let me know. I am devastated. :/

If you need any additional information from me, I'd be happy to provide. Thank you!

morrownr, 2024-09-07:
Hi @c0ncatenate 

Please run the following command and post the result:

$ lsusb

c0ncatenate, 2024-09-07:
$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
Bus 001 Device 004: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 001 Device 005: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

c0ncatenate, 2024-09-07:
An update for you @morrownr 

I tried the [RTW88 Testing Driver](https://github.com/lwfinger/rtw88) and it works alright for me. :D

Thank you for your continued efforts in updating the features. I am pleased to report that all the features are functioning as intended and I appreciate your dedication to improving it and keeping up with the updates.

morrownr, 2024-09-09:
Thanks for the update. Glad you are going with rtw88. Hopefully the new rtl8821au and rtl8812au drivers that are in the rtw88 repo make it into the kernel soon. They are standards compliant and are in the process of being accepted into the kernel.

c0ncatenate, 2024-09-09:
Awesome! I'll be watching for those updates. It's great to see that the rtl8821au and rtl8812au drivers are getting closer to kernel integration. Thanks for keeping the community informed and for all the work you put into this!

closed 2024-09-11 by c0ncatenate

morrownr, 2024-09-11:
Thanks for the kind words.