morrownr/8821au-20210708 #152: Building issue on rPi OS 12 aarch64

stouracp, 2024-09-17:
Please make sure, the script works on rPi OS 12 64bit, the workaround I made is, to build the module on rPi OS 11 32bit.
The issue is, it cannot locate properly kernel-headers.

morrownr, 2024-09-17:
Hi @stouracp 

Did you read FAQ.md? Look at the last item.

FYI: This is a problem with RasPiOS, not the script in this driver.

stouracp, 2024-09-18:
There is answer for rPi 4/400, not for Zero 2W (would be irrelevant) but more important, nothing regarding 11/12 version. Still running this on 12 64-bit is not working, so, what would be a workaround, to compile the kernel module on 12 64-bit vrsion?

morrownr, 2024-09-18:
> There is answer for rPi 4/400, not for Zero 2W...

I'll look at rewording the issue so that it says any 64 bit RasPi.

> Please make sure, the script works on rPi OS 12 64bit,

Can you be more specific about what problem you are seeing? I'm guessing because you provided basically nothing to go on.

stouracp, 2024-09-19:
interesting, no idea what have changedm, but today the compiling was succesful.

morrownr, 2024-09-19:
Is the adapter working?

FYI: There should soon by in-kernel support for your 8811au adapter. I have been helping the primary developer and the driver has been submitted to linux-wireless. It is under review currently.

stouracp, 2024-09-20:
Yes, the adapter is working, well, I have two in Zero 2W. Both are working.