morrownr/8821au-20210708 #153: Unable to install driver on a RPI 5 on version 6.1.77-v8+.

Wootter, 2024-09-22:
So, when I run the download command I receive:

: ---------------------------
: install-driver.sh v20240314
: aarch64 (kernel architecture)
: arm64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 8133036 (total system memory)
: 6.1.77-v8+ (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (Debian 14.2.0-3) 14.2.0
: dkms-3.0.13
: This system doesn't support Secure Boot
: ---------------------------

Checking for previously installed drivers...
Deleting module rtl8821au-5.12.5.2 completely from the DKMS tree.
Removing 8821au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8821au-5.12.5.2
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------

The kernel is built without module signing facility, modules won't be signed

Building module:
Cleaning build area...
Building module(s)...(bad exit status: 2)
Failed command:
kernelver=6.1.77-v8+ ./dkms-make.sh
Error! Bad return status for module build on kernel: 6.1.77-v8+ (aarch64)
Consult /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

It says my system doesn't support secure boot, but i'm confused on how to resolve the issue. Can anyone help me find a solution?

Adapter: ALFA AWUS036ACS
Device: Raspberry pi5
Version: 6.1.77-v8+

morrownr, 2024-09-22:
Hi @Wootter 

You did not mention what distro you are using. Can you elaborate?

Wootter, 2024-09-23:
Hello @morrownr 

Forgot, my apologies. I run Kali Linux, I follow every step that was listed, and even a few tutorials but nothing helped when i received that error.

Wootter, 2024-09-28:
Do you have any suggestions on how to fix this?

morrownr, 2024-09-28:
@Wootter 

Sorry for the delay. I am having some health problems that are slowing me down.

I suspect the issue is that the header files are not installed. It seems each distro uses a different name for the header files and Kali for x64 uses a different name than the distro for ARM64. Find the name of the package to install the header files and it should work. If you find it, let me know the name and I will add it to the docs.

Wootter, 2024-09-29:
@morrownr 

My bad, didn't know you're going through that. Hope you recover fast from your health problems! I'll check out which header files I need to download so it works. Thanks for the reply :)

morrownr, 2024-09-29:
No problem. Let me know what you discover.

FYI: At some point this year, we should see a good in-kernel driver make its way into the kernel for this chip. The driver is in review with the appropriate kernel devs so this is not a dream, it is happening.

Wootter, 2024-09-30:
@morrownr 

You're the best, i searched which header i needed and ended up on -> https://www.kali.org/docs/arm/raspberry-pi-5/ which helped me.

I did: 

cd /usr/src/kernel
sudo git clean -fdx && sudo make bcm2711_defconfig && sudo make modules_prepare

and afterwards 

sudo apt install --no-install-recommends realtek-rtl88xxau-dkms

Then i went to reinstall it, didn't edit the files, and rebooted and dang that worked flawlessly. Finally can start utilizing wifite:)