morrownr/8821au-20210708 #154: Edimax AC600 can not work with 5G

KES777, 2024-09-29:
sdss

KES777, 2024-09-29:
If I add this option `options 8821au rtw_wireless_mode=64` and reboot then WIFI routed is not listed:
[image] I expect to see BELL575 here.

morrownr, 2024-09-29:
Hi @KES777 

The first thing I think of when I see a report where 2g is working and 5g is not is that something is amiss with the regulatory information. What is your country?

Report the results of the following:

$ iw reg get

Remember to return any settings that you try to what they were if they don't work or we will end up self inflicting a wound.

KES777, 2024-09-30:
I bought this device in Poland, then moved into Canada.

```
iw reg get
global
country CA: DFS-FCC
	(2402 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
	(5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR
```
Though this device worked with TP-Link in 5G mode. Let me share these device models tomorrow.

KES777, 2024-10-13:
APs I use to connect to:
TL-WPA8631P - I am able to connect to 5G via this device.
Sagemcom GigaHub (FAST 5689E). - I am not able to connect to 5G via this device. I am able to connect to 5G via phone.

morrownr, 2024-10-13:
@KES777 

Hmmm... so you are only seeing the problem with the Sagemcom GigaHub AP? What channels are the two APs using?

If one is using a lower channel like 36 and the other is using a higher channel like 149, you might switch them and see if that makes a difference.

KES777, 2024-10-14:
I do not know. If I could find any instructions how to switch that I could test that.