morrownr/8821au-20210708 #155: 5Ghz networks not working properly

619tmFALCON, 2024-10-18:
I am using a TP-Link AC600 RTL8811AU on Zorin OS Core 17.2, and only 2.4Ghz networks show up. The weird thing is it used to work properly, but I had dual boot and I wanted to get rid of Windows, so I formatted (I was running Zorin OS Core 17.1 before btw). I also changed my processor from Ryzen 3200g to Ryzen 5700x3d. When I reinstalled the OS, it stopped showing 5Ghz networks. I tried changing the router settings to the recommended, and sometimes the network does appear, but I've only managed to connect to it once, and it was running at really slow speeds (20Mbps peak).

This is what I got when I ran the command:

```
6.8.0-47-generic x86_64
SecureBoot disabled
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2357:0604 TP-Link TP-Link UB500 Adapter
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1532:008a Razer USA, Ltd RZ01-0325, Gaming Mouse [Viper Mini]
Bus 001 Device 007: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
Bus 001 Device 005: ID 352d:2383 Drunkdeer Drunkdeer A75 ANSI
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
3: phy3: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8821au/5.12.5.2, 6.8.0-47-generic, x86_64: installed
phy#3
	Interface wlx3c52a1af4c80
		ifindex 5
		wdev 0x300000001
		addr 3c:52:a1:af:4c:80
		type managed
		txpower -100.00 dBm
global
country AR: DFS-FCC
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
```

morrownr, 2024-10-18:
Buen Dia @619tmFALCON 

There is nothing obviously wrong in the information that you posted.

What kernel is Zorin using?

$ uname -r

> I tried changing the router settings to the recommended in the readme...

If you remember the router changes that you make and the before and after settings, can you post them?

> country AR: DFS-FCC

It looks to me like AR needs updates to the regulatory data for the 6 GHz band but that would not be causing this problem. It is just something that needs to be investigated and accomplished.

619tmFALCON, 2024-10-18:
> What kernel is Zorin using?

I ran uname -r (I think you put it there because that's the command to check the kernel, right?). The output was 6.8.0-47-generic.

> If you remember the router changes that you make and the before and after settings, can you post them?

Yeah, I used the WiFi Analyzer app and saw that there were 8 networks on channels 149-165, including mine. I set 20MHz fixed width (which was on auto) and moved it to channel 140.

> It looks to me like AR needs updates to the regulatory data for the 6 GHz band but that would not be causing this problem. 

Yeah, as far as I know there's no way to use 6GHz here.

morrownr, 2024-10-18:
> saw that there were 8 networks on channels 149-165...

That does not mean that there is a lot of congestion but 8 is certainly enough to cause a lot of congestion if they are heavily used.

> I set 20MHz fixed width...

Try 80 MHz fixed and channel 36.

> moved it to channel 140.

It looks like 140 is a DFS channel just like it is in my country. That is fine as long a something does not conflict but let's no go there until we see if things are working on okay on non-DFS channels.

619tmFALCON, 2024-10-18:
It doesn't seem to let me choose 80MHz fixed.
[image]

morrownr, 2024-10-18:
Use 20/40/80 for 5 GHz and 20 for 2.4 GHz.

Here is info on WiFi channels:
https://en.wikipedia.org/wiki/List_of_WLAN_channels

>  the lowest it lets me go is channel 52. I think I saw in the WiFi Analyzer that 36 was a 2.4GHz channel, 

Channels 1-13 are the 2.4 GHz WiFi channels. Basically channels 1-11 are pretty much universal around the world but some countries also use 12 and 13.

There is not really a universal set of channels for 5 GHz. 

> (5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW

The 5170 in the above line tells me channel 36 is authorized in your country. It is not clear to me why your router is not allowing it. Maybe your router could use a firmware upgrade?

Does the router use different names for the SSIDs?

619tmFALCON, 2024-10-18:
Understood. What channel should I try? It only lets me use 52, 100, 132 or 149 when using 20/40/80.

> Maybe your router could use a firmware upgrade?

Oh, definitely. It's been untouched for years. Although I don't know how to do it.

> Does the router use different names for the SSIDs?

Yes, same name but one ends in 2.4GHz and the other one in 5.8Ghz (I don't know why they call it that, it's actually 5GHz. Every single ISP in Argentina does it)

closed 2024-10-18 by 619tmFALCON