morrownr/8821au-20210708 #157: Low Speed and Frequent Disconnects with TP-Link Archer T3U AC600 on Arch Linux (Kernel 6.6.60-1-lts)

XSebastian-GimenezX, 2024-11-15:
Hello,

I am experiencing performance issues and frequent disconnects with my **TP-Link Archer T3U AC600** WiFi adapter on Arch Linux. The adapter works correctly on Windows, including 5GHz networks, but on Linux, I am seeing very low speeds and frequent disconnections, even on 2.4GHz networks.

**Environment Information:**

- Adapter: TP-Link Archer T3U AC600 (USB device ID: 2357:011f TP-Link 802.11ac WLAN Adapter)
- Kernel: 6.6.60-1-lts
- Distribution: Archcraft (version 2024.10.22)
- Driver: morrownr/8821au-20210708 (latest available version)

**Expected Behavior:**

I expect the WiFi adapter to function correctly, as it did with a previous driver in the past and as it does on Windows, with proper speeds and without frequent disconnects, on both 2.4GHz and 5GHz networks.
**Current Behavior:**

- Extremely low download speeds, not exceeding 1 Mbps on 2.4GHz networks.
- Frequent disconnects.
- Unable to connect to 5GHz networks (although I understand this may be due to limitations in the current driver).

**Logs and Troubleshooting:**

1.  Logs:

- I checked system logs (dmesg and journalctl), but I did not find significant errors related to the WiFi device. I can provide more detailed logs if needed.

2.  Additional Tests:

- Tried different USB ports and routers, but the issue persists.
- Other USB devices work fine.
-  Tested the adapter on Windows, and it works perfectly with normal speeds and stability.

The previous driver I was using worked perfectly, but it is no longer compatible with my current system. This is the only driver I have found that recognizes my adapter, but it is performing poorly as described.

**Request for Assistance:**

Any help or suggestions on how to improve the stability and performance would be greatly appreciated. If further information is needed (full logs, etc.), I’d be happy to provide it.

Thank you for your work and for taking the time to look into this issue.

morrownr, 2024-11-15:
Did you fully remove any previously installed drivers?

$ dkms status

XSebastian-GimenezX, 2024-11-15:
Actually now that I execute this command I am doubting that the two dkms drivers are from this driver

`8812au/5.6.4.2_35491.20191025, 6.6.61-1-lts, x86_64: installed
nvidia/565.57.01, 6.6.61-1-lts, x86_64: installed
rtl8821au/5.12.5.2, 6.6.61-1-lts, x86_64: installed`

XSebastian-GimenezX, 2024-11-15:
I already removed the previous Wi-Fi driver, the only thing that seems to be is that it increases the connection from a maximum of 1 MB to about 2 MB on average, then the rest seems the same

morrownr, 2024-11-15:
I am really busy right now. Let me suggest this:

My intention for this repo is to ensure it is working well with kernel 6.12 and then post a message that maintenance is being discontinued due to in-kernel support starting with kernel 6.13.

You can try the new in-kernel support by going to the following repo:

https://github.com/lwfinger/rtw88

Let me know what you think.

XSebastian-GimenezX, 2024-11-16:
Problem solved. With the repo it started reading 5ghz networks and after moving my wifi to the range 34 to 66 I was able to connect without problem

closed 2024-11-16 by XSebastian-GimenezX