morrownr/8821au-20210708 #158: Install works fine, get segfault when running

nohren, 2025-01-03:
[image]

On step 1, instead of looking for linux-headers which don't exist in kali raspberry pi distribution, I am using it like so.

`sudo apt install -y kalipi-kernel-headers build-essential bc dkms git libelf-dev rfkill iw`

Notice kalipi-kernel-headers

The driver builds just fine.  Of note, gcc --version came up with major version 14.  And kernel gcc was major version 11.  gcc-11 is not available to kali linux distro.  I tried changing gcc --version to 12 which is available.  Same issue.  So then I tried building gcc-11 from source, and that failed after a very lengthy process.  I feel I've run out of sane options.  Could use some guidance here.

morrownr, 2025-01-03:
Recommend you try the following driver:

https://github.com/lwfinger/rtw88

nohren, 2025-01-03:
@morrownr do you know what could be causing the segfault?  Is it most likely gcc version mismatch or something else?

morrownr, 2025-01-03:
> do you know what could be causing the segfault?

No

>  Is it most likely gcc version mismatch or something else?

No idea. 

Kali for RasPi's has been very problematic. You can install RasPiOS and then install the apps you want. I am not going to work any problems with Kali.