morrownr/8821au-20210708 #159: Build fails on CM3588 board (RockChip RK3588 based ARM board) with Debian Bookworm for a TP-Link T2U Nano (rtl8811au)

agusventuri, 2025-01-12:
Hi!

I'm trying to build drivers for a TP-Link t2u nano adapter that has the rtl8811au chipset but can't get it to work. I'm guessing the Debian  version I'm using has modifications that make it incompatible.
Here are some logs and info about the setup I have and the output I'm getting.

## Hardware
- [CM3588 NAS Kit](https://wiki.friendlyelec.com/wiki/index.php/CM3588#Hardware_Spec)
- [TP-Link Archer T2U Nano](https://www.tp-link.com/ar/home-networking/adapter/archer-t2u-nano/)

## Software
### OS info
```bash
root@CM3588-Plus:~# cat /etc/*-release
# PLEASE DO NOT EDIT THIS FILE
BOARD="CM3588-Plus"
BOARD_NAME="CM3588-Plus"
LINUXFAMILY=nanopi6
BRANCH=dev
BOARD_VENDOR=FriendlyELEC
ARCH=arm
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

### Kernel info
```bash
root@CM3588-Plus:~# uname -r
6.1.57
```

### USB device info
```bash
root@CM3588-Plus:~# lsusb
Bus 003 Device 002: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
```

### Compiler info
```bash
root@CM3588-Plus:~# cat /proc/version
Linux version 6.1.57 (fa@jensen) (aarch64-linux-gcc (ctng-1.25.0-119g-FA) 11.3.0, GNU ld (GNU Binutils) 2.38) #13 SMP Tue May 21 16:26:32 CST 2024

gcc --version
gcc (Debian 12.2.0-14) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

# Issue

When executing the `install-driver.sh` it fails. Here's the result:
```bash
root@CM3588-Plus:~# ./install-driver.sh
Error! Bad return status for module build on kernel: 6.1.57 (aarch64)
Consult /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 140.26 seconds
```

Exploring the logs this is what I'm getting:
```bash
root@CM3588-Plus:~# cat /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log
DKMS make.log for rtl8821au-5.12.5.2 for kernel 6.1.57 (aarch64)
[... 84 lines cut ...]
  471 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8821au/5.12.5.2/build/include/osdep_service_linux.h:98,
                 from /var/lib/dkms/rtl8821au/5.12.5.2/build/include/osdep_service.h:57,
                 from /var/lib/dkms/rtl8821au/5.12.5.2/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8501:6: note: declared here
 8501 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/ioctl_cfg80211.c:490:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  490 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/cfg80211.h:8484:6: note: declared here
 8484 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
At top level:
cc1: note: unrecognized command-line option ‘-Wno-enum-int-mismatch’ may have been intended to silence earlier diagnostics
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8821au/5.12.5.2/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2057: /var/lib/dkms/rtl8821au/5.12.5.2/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.57'
make: *** [Makefile:2498: modules] Error 2
```

I tried several things, including modifying the c files myself, but I'm out of my depth there. If there's anything else you need from logs or output please let me know.

morrownr, 2025-01-12:
Hi @agusventuri 

It probably would not take me too long to track that problem down but I'd prefer that you try a different driver:

https://github.com/lwfinger/rtw88

That is a driver that we are using for testing and the patching go upstream into the Linux kernel. In fact, the rtl8821/11au driver went into kernel 6.13. More patches to improve a few things will go into kernel 6.14. However, that repo has all work. We need it tested to see if any additional work needs to be done.

With my driver that you were trying to install:

$ make clean
$ sudo sh remove-driver.sh

That will clean things up and you can go to the rtw88 repo and give it a shot.

agusventuri, 2025-01-12:
Hi @morrownr . Thanks for the quick reply.

I removed your driver and tried with the rtw88 but couldn't get it to work. The driver builds correctly but can't make it work beyond that.
I'm guessing then this FAQ applies: [https://github.com/lwfinger/rtw88?tab=readme-ov-file#q1-my-driver-builds-and-loads-correctly-but-fails-to-work-properly-can-you-help-me](https://github.com/lwfinger/rtw88?tab=readme-ov-file#q1-my-driver-builds-and-loads-correctly-but-fails-to-work-properly-can-you-help-me) and I should not post a new issue in the rtw88 repo.

morrownr, 2025-01-12:
@agusventuri 

> I'm guessing then this FAQ applies:

No, it no longer applies. I'll try to fix that today. The original owner of the repo wrote that but it really does not apply anymore so please post an issue giving as much detail as you think will help figure out the problem.