morrownr/8821au-20210708 #160: Can not connect after successfully install

berndver, 2025-01-23:
2 months ago I set up a Fujitsu Futro and connected it to the Internet using the [WIFI USB stick](https://www.ebay.de/itm/364722636448). It worked without any problems. I didn’t have to install a driver. Plug & Play.

Yesterday I wanted to set up another Fujitsu Futro. So I bought exactly the same WIFI USB stick again. Yesterday I plugged in the USB stick and nothing happened. No WLAN networks were displayed. I found that strange. Then I tried to install this driver. Now networks were displayed. But I could only connect to my mobile hotspot. I couldn’t connect to my network router. In addition, the passwords are not saved and are gone after a restart.

What has changed in the meantime, assuming that the WIFI adapter is the same chip? Can I perhaps install an earlier Lubuntu version so that the stick works? Or are there other settings that affect this? Or change some settings?

The stick shows up as Realtek rtl8812au/rtl8821au.

Thank you very much for your help!

mikepurvis, 2025-01-23:
I'm getting a very stable connection using this driver on kernel 6.5.0-44 (mantic) and 6.8.0-51 (noble) with an Archer T2U Nano.

morrownr, 2025-01-23:
@berndver @mikepurvis

Starting with kernel 6.14 in Ubuntu 25.04, you won't need to use these out-of-kernel drivers anymore. A nice new standards compliant driver for this chip went into kernel 6.13 and additional performance patches are going into kernel 6.14. If you would like to get an early look, you can install the new driver directly from the repo where we have been working on it after uninstalling this driver:

https://github.com/lwfinger/rtw88

@berndver 

> What has changed in the meantime, assuming that the WIFI adapter is the same chip?

Let's make sure it is the same chip:

$ lsusb

mikepurvis, 2025-01-27:
That's awesome! Thank you for the work you've done maintaining this out of tree driver all these years, and now pushing for an even better solution for users— I definitely look forward to switching to 6.14 on my Noble install when it's available.