morrownr/8821au-20210708 #161: Build fails on kernel 6.13.1-arch1-1

tsukiraa, 2025-02-04:
Driver fails to build using dkms
/var/lib/dkms/rtl8821au/make.log:

```
Building module(s)
# command: kernelver=6.13.1-arch1-1 ./dkms-make.sh
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.13.1-arch1-1/build M=/var/lib/dkms/rtl8821au/5.12.5.2/build  modules
make[1]: Entering directory '/usr/lib/modules/6.13.1-arch1-1/build'
make[2]: Entering directory '/var/lib/dkms/rtl8821au/5.12.5.2/build'
  CC [M]  core/rtw_cmd.o
  CC [M]  core/rtw_security.o
  CC [M]  core/rtw_debug.o
  CC [M]  core/rtw_io.o
  CC [M]  core/rtw_ioctl_query.o
  CC [M]  core/rtw_ioctl_set.o
  CC [M]  core/rtw_ieee80211.o
  CC [M]  core/rtw_mlme.o
  CC [M]  core/rtw_mlme_ext.o
  CC [M]  core/rtw_mi.o
  CC [M]  core/rtw_wlan_util.o
  CC [M]  core/rtw_vht.o
core/rtw_mlme_ext.c: In function ‘OnAuth’:
core/rtw_mlme_ext.c:2568:41: warning: suggest braces around empty body in an ‘if’ statement [-Wempty-body]
 2568 |                                         ;/* TODO: STA re_auth within expire_to */
      |                                         ^
core/rtw_mlme_ext.c:2573:33: warning: suggest braces around empty body in an ‘if’ statement [-Wempty-body]
 2573 |                                 ; /* TODO: STA re_auth and auth timeout */
      |                                 ^
  CC [M]  core/rtw_pwrctrl.o
  CC [M]  core/rtw_rf.o
  CC [M]  core/rtw_chplan.o
  CC [M]  core/monitor/rtw_radiotap.o
  CC [M]  core/rtw_recv.o
  CC [M]  core/rtw_sta_mgt.o
  CC [M]  core/rtw_ap.o
  CC [M]  core/wds/rtw_wds.o
  CC [M]  core/mesh/rtw_mesh.o
  CC [M]  core/mesh/rtw_mesh_pathtbl.o
  CC [M]  core/mesh/rtw_mesh_hwmp.o
  CC [M]  core/rtw_xmit.o
  CC [M]  core/rtw_p2p.o
  CC [M]  core/rtw_rson.o
core/rtw_recv.c: In function ‘validate_recv_frame’:
core/rtw_recv.c:2479:79: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 2479 |                         DBG_COUNTER(adapter->rx_logs.core_rx_pre_data_handled);
      |                                                                               ^
  CC [M]  core/rtw_tdls.o
  CC [M]  core/rtw_br_ext.o
  CC [M]  core/rtw_iol.o
  CC [M]  core/rtw_sreset.o
  CC [M]  core/rtw_btcoex_wifionly.o
core/rtw_xmit.c: In function ‘update_attrib’:
core/rtw_xmit.c:1696:74: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 1696 |                 DBG_COUNTER(padapter->tx_logs.core_tx_upd_attrib_unknown);
      |                                                                          ^
  CC [M]  core/rtw_btcoex.o
  CC [M]  core/rtw_beamforming.o
  CC [M]  core/rtw_odm.o
  CC [M]  core/rtw_rm.o
  CC [M]  core/rtw_rm_fsm.o
  CC [M]  core/rtw_ft.o
  CC [M]  core/rtw_wnm.o
  CC [M]  core/rtw_mbo.o
  CC [M]  core/rtw_rm_util.o
  CC [M]  core/efuse/rtw_efuse.o
  CC [M]  core/rtw_roch.o
  CC [M]  core/crypto/aes-internal.o
  CC [M]  core/crypto/aes-internal-enc.o
  CC [M]  core/crypto/aes-gcm.o
  CC [M]  core/crypto/aes-ccm.o
  CC [M]  core/crypto/aes-omac1.o
  CC [M]  core/crypto/ccmp.o
  CC [M]  core/crypto/gcmp.o
  CC [M]  core/crypto/aes-siv.o
  CC [M]  core/crypto/aes-ctr.o
  CC [M]  core/crypto/sha256-internal.o
  CC [M]  core/crypto/sha256.o
  CC [M]  core/crypto/sha256-prf.o
  CC [M]  core/crypto/rtw_crypto_wrap.o
  CC [M]  core/rtw_swcrypto.o
  CC [M]  os_dep/osdep_service.o
  CC [M]  os_dep/linux/os_intfs.o
  CC [M]  os_dep/linux/usb_intf.o
  CC [M]  os_dep/linux/usb_ops_linux.o
  CC [M]  os_dep/linux/ioctl_linux.o
  CC [M]  os_dep/linux/xmit_linux.o
  CC [M]  os_dep/linux/mlme_linux.o
  CC [M]  os_dep/linux/recv_linux.o
  CC [M]  os_dep/linux/ioctl_cfg80211.o
  CC [M]  os_dep/linux/rtw_cfgvendor.o
  CC [M]  os_dep/linux/wifi_regd.o
  CC [M]  os_dep/linux/rtw_android.o
  CC [M]  os_dep/linux/rtw_proc.o
os_dep/linux/recv_linux.c: In function ‘rtw_os_recv_indicate_pkt’:
os_dep/linux/recv_linux.c:556:68: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
  556 |                         DBG_COUNTER(padapter->rx_logs.os_netif_err);
      |                                                                    ^
  CC [M]  os_dep/linux/nlrtw.o
  CC [M]  os_dep/linux/rtw_rhashtable.o
  CC [M]  hal/hal_intf.o
  CC [M]  hal/hal_com.o
  CC [M]  hal/hal_com_phycfg.o
  CC [M]  hal/hal_phy.o
os_dep/linux/ioctl_cfg80211.c:10523:32: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, struct cfg80211_chan_def *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct cfg80211_chan_def *)’ [-Wincompatible-pointer-types]
10523 |         .set_monitor_channel = cfg80211_rtw_set_monitor_channel,
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
os_dep/linux/ioctl_cfg80211.c:10523:32: note: (near initialization for ‘rtw_cfg80211_ops.set_monitor_channel’)
make[4]: *** [/usr/lib/modules/6.13.1-arch1-1/build/scripts/Makefile.build:194: os_dep/linux/ioctl_cfg80211.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/lib/modules/6.13.1-arch1-1/build/Makefile:1989: .] Error 2
make[2]: *** [/usr/lib/modules/6.13.1-arch1-1/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/rtl8821au/5.12.5.2/build'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.13.1-arch1-1/build'
make: *** [Makefile:2498: modules] Error 2

# exit code: 2
# elapsed time: 00:00:10
```

morrownr, 2025-02-04:
I'll set aside time to fix this today.

morrownr, 2025-02-04:
Patches merged. Try:

$ git pull

$ sudo sh install-driver.sh

iliqiliev, 2025-02-06:
This could be closed? It works fine now

morrownr, 2025-02-06:
Thanks for the report.

closed 2025-02-06 by morrownr