morrownr/8821au-20210708 #165: PROPER INSTALLATion follow my setps :-

gauravlinux, 2025-02-16:
sudo apt install linux-headers-$(uname -r)
git clone https://github.com/morrownr/8821au-20210708.git
cd 8821au-20210708
sudo apt install bc
sudo sh install-driver.sh
n
y

morrownr, 2025-02-16:
Hi @gauravlinux 

Is this a question? If so, I cannot tell you without knowing what distro you are using.

Slayerwen, 2025-03-03:
> Hi [@gauravlinux](https://github.com/gauravlinux)
> 
> Is this a question? If so, I cannot tell you without knowing what distro you are using.

I think he mean this is a working way to install the driver. The other way dont work in the guide

morrownr, 2025-03-03:
Hi @Slayerwen 

> The other way dont work in the guide...

The installation steps in the README work for many distros. Note that I took the time to give specific instructions for numerous distros. Unfortunately there are so many different distros that it is impossible to document all cases. It would be nice if there was more standardization. His information is of little value without knowing exactly what distro and version he is providing the information for.

However, a new, in-kernel driver went into kernel 6.14 so it should not be long that this driver does not matter. If fact, you can install the new driver now from the repo where we developed it:

https://github.com/lwfinger/rtw88

The quality of the new driver is much better than this Realtek driver.

Slayerwen, 2025-03-03:
> Hi [@Slayerwen](https://github.com/Slayerwen)
>
>
> The installation steps in the README work for many distros. Note that I took the time to give specific instructions for numerous distros. [...]

Thank you! I will try the new driver now :)