morrownr/8821au-20210708 #168: Raspberry Pi 5 Kali Linux Archer T2U PLUS(8821au)

JustinSS23, 2025-02-22:
: ---------------------------
: install-driver.sh v20241003
: aarch64 (kernel architecture)
: arm64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 2026304 (total system memory)
: 6.6.63-v8+ (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (Debian 14.2.0-16) 14.2.0
: dkms-3.1.5
: This system doesn't support Secure Boot
: ---------------------------

Checking for previously installed drivers...
Removing a driver that was added to dkms.
Deleting module rtl8821au/5.12.5.2 completely from the DKMS tree.
Removing 8821au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8821au-5.12.5.2
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------

The kernel is built without module signing facility, modules won't be signed

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
kernelver=6.6.63-v8+ ./dkms-make.sh

Error! Bad return status for module build on kernel: 6.6.63-v8+ (aarch64)
Consult /var/lib/dkms/rtl8821au/5.12.5.2/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.

Thought I followed the instructions pretty well, updated/upgraded everything but I am stuck getting this error.

morrownr, 2025-02-22:
Hi @JustinSS23 

This is likely a problem with the header files not being installed.

> Thought I followed the instructions pretty well...

You may have followed the instructions well but it is impossible to fully document all of the possible options in this day in time.

Tell me what Linux distro you are using.

Also, let me point you to a repo where we have been working on a new rtl8821/11au driver:

https://github.com/lwfinger/rtw88

That repo is where we have been working on the in-kernel rtw88 drivers. New drivers for the rtl8812au and rtl8821/11au chips were upstreamed and added to the Linux mainline kernel recently. The drivers are fully in the kernel with kernel 6.14. Since you are using 6.6, you can install the driver from https://github.com/lwfinger/rtw88 . If you have problems, post your issues in the rtw88 repo.

Remember to uninstall the driver in this repo:

$ sudo sh remove-driver.sh

Also, the Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Reading menu items 1 and 2 can be helpful.

Cheers

JustinSS23, 2025-02-22:
I am running Kali Linux but as you said it looks like my headers are missing which looks to be stemming from using an older kernel version. I am attempting a full-upgrade now to see if that will make any changes.

JustinSS23, 2025-02-23:
Noted, will definitely do that instead.