morrownr/8821au-20210708 #172: 5 GHz not working on Ubuntu.

ubarbosa, 2025-03-12:
ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]

The 2.4 GHz network is working normally. However, the 5 GHz network does not connect.
So, I am facing the following problem:
Since the access provider does not provide the router password, I cannot change its configuration. It happens that, approximately every 10 minutes, it detects that the 5 GHz signal from the adapter is strong and disconnects, hoping that the system will make a new 5 GHz connection.

I mean, every 10 minutes, the following message is displayed:

wlxc006c3914137: CTRL-EVENT-DISCONNECTED bssid=c8:9f:1a:d7:4f:00 reason=0

And then, the whole reconnection process takes place.

I have already configured NetworkManager to only connect to 2.4 GHz, but this does not prevent my connection from dropping every 10 minutes and reconnecting again.

I have some programming experience, but practically none in drivers.
Even so, it doesn't hurt to ask.

Is there any way to turn off the 5GHz signal in the driver?
Even if it has to be done by changing one or more of the source code, could you at least tell me which source programs I should study?

morrownr, 2025-03-12:
> Is there any way to turn off the 5GHz signal in the driver?

Yes. 

$ ./edit-options:

Add the `rtw_wireless_mode` parameter to the `options` line as shown:

options 8821au rtw_led_ctrl=1 rtw_wireless_mode=11

ubarbosa, 2025-03-14:
No programming is necessary.
You had already coded this feature.

morrownr, 2025-03-14:
Well, actually Realtek coded the feature but I had to read the code to document it.