morrownr/8821au-20210708 #173: need help

Hussien110, 2025-04-01:
: ---------------------------
: install-driver.sh v20241003
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 16/16 (in-use/total processing units)
: 15535204 (total system memory)
: 6.12.13-amd64 (kernel version)
: gcc-14 (version of gcc used to compile the kernel)
: gcc (Debian 14.2.0-19) 14.2.0
: dkms-3.1.5
: mokutil not installed (Secure Boot status unknown)
: ---------------------------

Checking for previously installed drivers...
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6-23
Creating symlink /var/lib/dkms/rtl8812au/5.13.6-23/source -> /usr/src/rtl8812au-5.13.6-23
The driver was added to dkms successfully.
: ---------------------------

Error! Your kernel headers for kernel 6.12.13-amd64 cannot be found at /lib/modules/6.12.13-amd64/build or /lib/modules/6.12.13-amd64/source.
Please install the linux-headers-6.12.13-amd64 package or use the --kernelsourcedir option to tell DKMS where it's located.
An error occurred. dkms build error:  21
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

I'm using a newly installed Kali Linux, sorry but I'm new

morrownr, 2025-04-02:
Hi @Hussien110 

Sorry for the delay. I have been on the road.

Let me recommend this:

Run:

$ sudo ./remove-driver.sh

...to clear out this driver. Go to the following repo and install the driver there:

https://github.com/lwfinger/rtw88

That repo is where we do the work to be upstreamed into mainline.

You may still have an issue but we can handle it in that repo.

Hussien110, 2025-04-02:
I actually found a solution for that problem but I’m facing another issue
with the monitor mode because I didn’t edit the script and when I tried
your monitor mode repo it still didn't work, can you please help me with
the monitor mode issue and tell me what info you need, and thanks for the
help

morrownr, 2025-04-02:
I am a little confused. This repo is for rtl8821/11au chips and the driver you installed is for the rtl8812au chip. I need to know what you have. Can you run and post the results of:

$ lsusb

Hussien110, 2025-04-03:
Super soory for the late reply i was at work, I have an alfa wifi adapter model awus036ac and the output of lsusb is : 
lsusb                   
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 30c9:0065 Luxvisions Innotech Limited HP Wide Vision HD Camera
Bus 001 Device 003: ID 0489:e0e0 Foxconn / Hon Hai Wireless_Device
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

morrownr, 2025-04-03:
> ID 0bda:8812 

That is a rtl8812au based adapter. So you appear to have installed the correct driver but are now posting in a repo for a different chip.

Nevertheless, I am going to suggest you uninstall the driver you installed:

$ sudo sh remove-driver.sh

and install the following driver as it is more modern and will help you get used to the new in-kernel driver that went into kernel 6.14:

https://github.com/lwfinger/rtw88

I'll be at that repo also in case you need help.