morrownr/8821au-20210708 #174: Cant get 5ghz working

Connorplayer123, 2025-04-07:
Im using a TP-Link Archer T2U PLUS [RTL8821AU] on Debian Bookworm

I need help getting 5ghz working

morrownr, 2025-04-08:
Hi @Connorplayer123 

Probably an issue with the country code setting. Run and post the results of:

$ iw reg get

mholl1983, 2025-06-08:
@Connorplayer123 Did you ever get it working? I'm having the same issue.

morrownr, 2025-06-09:
Hi @mholl1983 

I can probably help. What kernel are you using?

mholl1983, 2025-06-09:
Thank you, @morrownr! Reinstalling the driver as per [this comment](https://github.com/morrownr/8821au-20210708/issues/180#issuecomment-2954321489) seemed to sort me, but just in case, I checked my kernel and it's `6.8.0-60-generic`.

mholl1983, 2025-06-10:
Well, that was shortlived. It disappeared again.

morrownr, 2025-06-11:
Are you using Ubuntu? If so, upgrading to 25.04 and you won't need to install a driver. Just remember to uninstall any drivers you have installed.

If that is not an option, you can install the new in-kernel driver on older kernels like yours by going to the following repo:

https://github.com/lwfinger/rtw88

Remember to uninstall any previously installed drivers.

morrownr, 2025-06-11:
@mholl1983 

> Is this rtw88 driver newer than the one I might have installed?

Yes, several years newer. In fact, it recently went into mainline with kernel 6.14. The most important thing about this new driver is that it is Linux Wireless Standards (mac80211) compliant. mac80211 is the modern standard to which wireless drivers should be written. This new standard has been in effect for years but Realtek has failed to update its drivers for USB to this standard. Unlike Mediatek, that has been following Linux Wireless Standards for years, Realtek has not been doing so. In fact, the new series of drivers is a community project but the drivers are excellent and it appears Realtek is taking over maintenance now that the drivers are in the kernel.

> I'm on Zorin OS which I believe is based on an LTS version of Ubuntu, but not 100% sure.

Zorin is based on the last LTS version of Ubuntu. 

Back to the original topic: If you are having problem with band 2 (5 GHz) only, the problem could be with wireless regulation. To see if your system is using the proper country code, you can run and paste the results of the following:

$ iw reg get

FYI: When I speak of rtw88, this is a series of drivers to support Realtek WiFi 5 chips. There is also rtw89 which is also a series of drivers to support Realtek WiFi 6 and 7 chips. The USB support for rtw88 is in the kernel with the latest kernels. USB support for rtw89 is in work. I am encouraging everyone that uses the Realtek out-of-kernel drivers here at my site to move to using the new drivers that can be installed on older kernels. Once Zorin is based on Ubuntu 26.04 LTS, you will no longer need to install a driver for your usb adapter.

mholl1983, 2025-06-24:
Ok, 5 ghz is disappearing again. I am ready to try the newer driver. Do I need to uninstall my existing one first? I understand the install steps, but not exactly sure of where the uninstall step should occur. Any help is appreciated.

morrownr, 2025-06-25:
> Do I need to uninstall my existing one first?

If the usb adapter is your only source of wifi then the answer is no. You can download the new driver from the following site per the instructions at that site. Remember that main reason for that site is development and testing so no fancy script to do everything for you... at least not yet

After you download the source for the new driver, you can then run the following to uninstall the driver in this repo:

$ sudo sh remove-driver.sh

Once the uninstallation is complete and you reboot, you can install the driver from the rtw88 site:

https://github.com/lwfinger/rtw88

If you have any problems with the new driver, you can report them at the rtw88 repo.

mholl1983, 2025-06-26:
Thank you! I *think* I'm sorted now, but I'm posting the results of `sudo lshw -C network`, just in case:

```
  *-network
       description: Wireless interface
       physical id: 4
       bus info: usb@2:3
       logical name: wlx984827de9409
       serial: 98:48:27:de:94:09
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtw_8821au driverversion=6.8.0-60-generic firmware=N/A ip=xx.xx.xx.xx link=yes multicast=yes wireless=IEEE 802.11
```

morrownr, 2025-06-26:
@mholl1983 

I do not see anything unusual so unless you see a problem, you should be in good shape. The rtw88 repo has the code that went into kernel 6.14 so you can get used to the new driver. Keep in mind that once you get to kernel 6.14, as long as your distro is properly installing the correct firmware file, you should be in plug and play heaven. I'm going to forecast that by mid-2026, the Linux kernel will have good in-kernel drivers for basically ALL modern WiFi 5 and later USB WiFi chips. Pretty cool huh?