morrownr/8821au-20210708 #175: Fedora 42 kernel 6.14.2

insel-maz, 2025-04-17:
I needed to pull the latest version of this driver to compile with Fedora 42 kernel 6.14.2.
According to the README, this driver is no longer needed since the other new driver is shipped with the kernel.

Thank you for providing a wifi driver that had served me faithfully over the years.

May I buy you a coffee this time?

onestan, 2025-04-24:
I couldn't get my TP-Link T2U Plus to work on the 6.14 kernel without installing the driver manually. 

Does anybody else have a similar issue?

morrownr, 2025-04-29:
No need for the coffee. Just stop by and say hi from time to time.

tetsuo76, 2025-04-30:
> I couldn't get my TP-Link T2U Plus to work on the 6.14 kernel without installing the driver manually.
> 
> Does anybody else have a similar issue?

I have the same USB wifi card and I couldn't get it to work using the latest kernel driver on fedora (6.14.4-200.fc41.x86_64)

morrownr, 2025-04-30:
Hi @onestan @tetsuo76 

Please remember that I help people with hundreds of different combinations of adapters/modules, distros, CPUs and configurations. 

I need to know what adapter you are using and what chipset is in it?

What exact distros and you running and what kernel on your using?

What CPU is in the system?

What should be happening and what appears to be the problem?

Both of you appear to be on kernel 6.14. With kernel 6.14, you should not need to install a driver if your adapter has a rtl8821/11au chip. With that said, it is possible that your distro has not recognized the new in-kernel driver and has not setup for it. How about we check to see if your distro has the proper firmware installed:

Post the results of:

$ ls -l /lib/firmware/rtw88

tetsuo76, 2025-05-02:
Hello morrownr and thanks for the reply. 

My wifi usb adapter is a TP-Link AC600 Archer T2U Plus and my CPU is an Intel i7-2600

https://www.tp-link.com/en/home-networking/high-gain-adapter/archer-t2u-plus/

My distro is an updated Fedora 41 (6.14.4-200.fc41.x86_64)

According to `lsusb` my chipset is RTL8821AU:

`Bus 002 Device 006: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]`

The results of `ls -l /lib/firmware/rtw88`:

```
-rw-r--r--. 1 root root   700 Apr 10 03:00 README.xz
-rw-r--r--. 1 root root 12648 Apr 30 16:37 rtw8703b_fw.bin.xz
-rw-r--r--. 1 root root 14080 Apr 30 16:37 rtw8703b_wow_fw.bin.xz
-rw-r--r--. 1 root root 17196 Apr 30 16:37 rtw8723d_fw.bin.xz
-rw-r--r--. 1 root root 16148 Apr 30 16:37 rtw8812a_fw.bin.xz
-rw-r--r--. 1 root root 33036 Apr 30 16:37 rtw8814a_fw.bin.xz
-rw-r--r--. 1 root root 18816 Apr 30 16:37 rtw8821a_fw.bin.xz
-rw-r--r--. 1 root root 48736 Apr 30 16:37 rtw8821c_fw.bin.xz
-rw-r--r--. 1 root root 74356 Apr 30 16:37 rtw8822b_fw.bin.xz
-rw-r--r--. 1 root root 77428 Apr 30 16:37 rtw8822c_fw.bin.xz
-rw-r--r--. 1 root root 57492 Apr 30 16:37 rtw8822c_wow_fw.bin.xz
```

and the results of dmesg after inserting the card:

[   44.040217] rtw_8821au 2-1.8:1.0: Firmware version 42.4.0, H2C version 0
[   47.265789] rtw_8821au 2-1.8:1.0: failed to poll offset=0x5 mask=0x1 value=0x0
[   47.265797] rtw_8821au 2-1.8:1.0: mac power on failed
[   47.265799] rtw_8821au 2-1.8:1.0: failed to power on mac
[   47.265800] rtw_8821au 2-1.8:1.0: failed to setup chip efuse info
[   47.265802] rtw_8821au 2-1.8:1.0: failed to setup chip information
[   47.265882] rtw_8821au 2-1.8:1.0: probe with driver rtw_8821au failed with error -16
[   47.265935] usbcore: registered new interface driver rtw_8821au

morrownr, 2025-05-02:
> -rw-r--r--. 1 root root 18816 Apr 30 16:37 rtw8821a_fw.bin.xz

That is the firmware that is required for rtl8821/11au chips. So it looks like Fedora has the firmware in place.

> [ 44.040217] rtw_8821au 2-1.8:1.0: Firmware version 42.4.0, H2C version 0

`rtw_8821au` tell me that you have installed rtw88 from: https://github.com/lwfinger/rtw88

The driver name in the kernel is `rtw88_8821au`.

What I think is happening here is that we are confusing things and too many things are installed... and they are conflicting with each other. How many drivers have you installed?

tetsuo76, 2025-05-02:
> What I think is happening here is that we are confusing things and too many things are installed... and they are conflicting with each other. How many drivers have you installed?

Thanks for the quick response. You are probably right. I was using https://github.com/morrownr/8821au-20210708 without any problems (5G/~350mbit) for a few months until some recent kernel upgrade (probably 6.12). I've also tried https://github.com/lwfinger/rtw88 since then but without any luck. I am almost certain that I uninstalled the drivers each time. Any idea how can I fix this mess?

morrownr, 2025-05-02:
@tetsuo76 

> Any idea how can I fix this mess?

I think we can clean things up.

> I was using https://github.com/morrownr/8821au-20210708

To uninstall that driver, run the following from the driver directory:

$ sudo sh remove-driver.sh

To uninstall the out-of-kernel version of rtw88, run the following from the rtw88 directory:

$ sudo make uninstall

Now let's see what we are. Run and post the results of the following:

$ dkms status

Now reboot

With the adapter in a port, run and post the results of:

$ lsmod | grep 8821

tetsuo76, 2025-05-02:
```
$ lsmod | grep 8821
rtw88_8821au           12288  0
rtw88_8821a            45056  1 rtw88_8821au
rtw88_88xxa            49152  1 rtw88_8821a
rtw88_usb              36864  1 rtw88_8821au
rtw88_core            319488  3 rtw88_88xxa,rtw88_8821a,rtw88_usb
```

morrownr, 2025-05-03:
> rtw88_8821au

You are now running with the in-kernel driver. Is it working?

If you have problems with it, report the problem to:

https://github.com/lwfinger/rtw88

onestan, 2025-05-03:
> [@tetsuo76](https://github.com/tetsuo76)
>
>
> I think we can clean things up.
>
>
> To uninstall that driver, run the following from the driver directory:
>
> $ sudo sh remove-driver.sh
>
> To uninstall the [...]

Hi, @morrownr! I'm sorry I couldn't give you more information earlier, but @tetsuo76 gave almost identical information as mine with the exception of my CPU model.

I followed the quoted steps and now the device works as expected! Thank you very very much! I really appreciate your work!

morrownr, 2025-05-03:
@onestan 

>  I'm sorry I couldn't give you more information earlier

No problem. Sometimes it takes someone else reporting the same thing with slightly different details for me to figure out what the problem may be, 

> now the device works as expected! Thank you very very much! I really appreciate your work!

You are welcome. If you see any bugs, the place to reports them so that the fixes work their way to the mainline kernel is:

https://github.com/lwfinger/rtw88

Just make sure your bug report say whether you are using the driver from the kernel or the out-of-kernel version in that repo.

tetsuo76, 2025-05-17:
@morrownr Many thanks for your help. I am on latest kernel now (6.14.6-200.fc41.x86_64) and my TP-Link T2U Plus is working great!

morrownr, 2025-05-17:
Glad to hear that.

shkabo, 2025-05-20:
I'm here just to say thank you @morrownr for the driver and also for assistance in this tread to disable this driver and continue using new in kernel driver.
I faced an issue where I couldn't upgrade my Fedora 40 to 6.14.5 kernel. Update went smooth via Software Center, rebooted PC, tried to boot to  6.14.5 kernel but ended up with a message `error ../../grub-core/fs/fshelp.c:257: file 'initramfs-6.14.5-100.fc40.x86_64.img' not found`. So after a bit of searching, brief chat with chatgpt, I ended up here where I found the actual solution `$ sudo sh remove-driver.sh` 

Thanks once again for your hard work

morrownr, 2025-05-20:
@shkabo 

Thank you for the kind words. The transition from the old drivers to the new ones will take a while for the overall community but the results should be good. There is now an effort to do the same thing with the rtw89 series of drivers but I am afraid this may be more problematic as almost all of the Realtek WiFi 6 and 7 adapters are multi-state adapters with Windows drivers onboard so we will have to see where that goes.