morrownr/8821au-20210708 #176: No issues yet

LAP47, 2025-04-24:
How do I message you on here without reporting an issue? I just wanted to let you know that this is working on Debian 12.10 KDE with a Netgear A6100 Wireless Adapter. Bus 002 Device 002: ID 0846:9052 NetGear, Inc. A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU].
6.1.0-33-amd64 and linux-headers-6.1.0-33-amd64,linux-headers-6.1.0-33-common. I don't know what else you need to know. 
https://www.netgear.com/home/wifi/adapters/a6100/.

morrownr, 2025-04-24:
Hi @LAP47 

> How do I message you on here without reporting an issue?

Reporting an issue is just fine. It is nice to get good reports.

Would you like to test something for me?

LAP47, 2025-04-25:
Depends on what it is. Or if I'm able to.

LAP47, 2025-04-25:
How do you start to build/make or develop or whatever the right word is the drivers for a USB dongle? Do you actually have to have the device to do it? I have three more NetGears (WG111 v2 and WG111 v3) an LG (AN-WF100) and a Belkin (F9L1002 v1).

morrownr, 2025-04-25:
@LAP47 

> Just a Linux user/tinkerer.

That works.

If you could uninstall this driver...

$ sudo sh remove-driver.sh

And then go and install the following driver...

https://github.com/lwfinger/rtw88

...that would be great. Take notes, complain about the README or anything that comes up.

That is a repo where we work on the new standards compliant driver. It has actually gone into kernel 6.14 but since you are not using kernel 6.14, you can install it from the above repo. One of these days where you get to kernel 6.14 or later, there will be no need to install anything, your adapter will just work. Cool huh?

morrownr, 2025-04-25:
> Is rtw88 replacing this driver?

Yes. I am not taking this driver down but I will not be updating it beyond support for kernel 6.14 as anyone that uses kernel 6.14 or later will not need it.

The rtl8811au driver in rtw88 is already in the mainline linux kernel as of kernel 6.14. We are maintaining this repo as a place for users on older kernels and as a place to test and fix things. This driver is made the way Linux driver should be made... that is why it was allowed into the kernel. The Realtek drivers do not meet standards and will not be allowed in the kernel.

> Does it matter if it's older or newer hardware?

No. I have systems in use here that are from 15 to 1 year old. My main dev box is 12 years old.

LAP47, 2025-04-25:
Okay. Do I have to reboot the system before moving onto step iv? Is this the only driver that rtw88 will be replacing? Because I'm also using another one of your drivers for my Linksys WUSB6300.

LAP47, 2025-04-25:
I don't know if I've got secure boot on or not ( I don't remember). I already installed dkms, though and finished step 3 so I just need to know if I should reboot before moving onto to step 4.

morrownr, 2025-04-25:
I think you rebooted before step iv so I'm going to say no. The dkms instructions were added recently by another person. If that does not work then I test it and we can make some changes. As you know, I like a single script to do the installation but that is not my repo so I tread lightly.

Remember to uninstall that other out-of-kernel driver you have installed.

LAP47, 2025-04-25:
I didn't reboot because I was waiting for confirmation/clarification first. No out of kernel drivers have been installed on this laptop yet. So I'm still at 2. iv. My mistake.

LAP47, 2025-05-03:
How do I uninstall rtw88? I don't know why but it's not working anymore.

morrownr, 2025-05-08:
If you installed it using the manual method:

$ sudo make uninstall

closed 2025-06-11 by morrownr