morrownr/8821au-20210708 #177: The new in-kernel driver being a "much better driver" as mentioned in the readme is not true in my experience

CapsAdmin, 2025-04-26:
I don't know about better code, but at least when it comes to user experience, the one in the kernel is causing system stability issues when scanning. Locking up the system so that you'd need to reboot and connection randomly dropping, the usb wifi dongle seem to just crash with the only way being to reconnect the device or reboot.

However, with this specific driver I never experienced any of these problems.

I tried switching to an rtl8821ae (some pci express variant) which doesn't work with this driver (afaik) but it was even worse. I ended up buying an intel pci variant which worked out of box, because online sentiment about realtek wifi drivers on linux is not very good.

LAP47, 2025-04-26:
So you prefer the one that you'd have to install manually?

morrownr, 2025-04-26:
@CapsAdmin 

It is very hard for me to understand the details of your experience. 

Exactly what wireless device is the subject concerning? This repo for a out-of-kernel Realtek driver for the rtl8821/11au chips as used in USB WiFi adapters and modules. However....

You mention "I tried switching to an rtl8821ae (some pci express variant) which doesn't work with this driver (afaik) but it was even worse." which is a totally different chips for use in pci applications.

Did you install the rtw88 driver as shown in " https://github.com/lwfinger/rtw88 " or are you using a distro with kernel 6.14?

Please clarify some of the issues.

closed 2025-06-11 by morrownr