morrownr/8821au-20210708 #178: Driver doesnt work on kernel 6.12 kali linux

drifter-08, 2025-05-10:
I've followed all the steps and installed the driver but when I run iwconfig or lsusb it doesnt show my wifi adapter
Kernel version 6.12.25, kali linux on a virtual machine
Is there any solution to this?

morrownr, 2025-05-11:
Hi @drifter-08 

> kali linux on a virtual machine

The following is the Main Menu for this site.

https://github.com/morrownr/USB-WiFi

If you scroll down to menu item 9, under it you will see a guide called `Using a USB Adapter in VirtualBox`. You did not say your were using VirtualBox but that is the guide that I have.

You also did not say what version of Kali, Is it for x86_64 or ARM64?

morrownr, 2025-06-11:
I am going to assume this issue has been resolved so I am closing the issue.

closed 2025-06-11 by morrownr