morrownr/8821au-20210708 #179: Secure Boot

StarMusa88, 2025-05-15:
Hi,

Great Work Much Appreciated, I just have 1 issue I'm currently Dual Booting from my Laptop from 2 different drives. 1 is windows 11 and 2 is Ubuntu, When I disable Secure Boot everything works fine but once Secure Boot is enabled the Wifi Adapter doesn't turn on. but shows in lsusb ( Bus 001 Device 012: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]

Thanks.

morrownr, 2025-05-16:
Which version of Ubuntu are you using?

Was secure boot on when you installed Ubuntu?

StarMusa88, 2025-05-18:
Hi,
Apologies for the late reply. I'm running on Ubuntu 24.04.2 LTS and Yes secure boot was enabled.

morrownr, 2025-05-19:
If you upgrade to Ubuntu 25.04, you won't need this driver at all as the new in-kernel driver works well. If you upgrade you installation, it would be best to uninstall this driver before you proceed if you have another source of internet.

closed 2025-05-19 by StarMusa88