morrownr/8821au-20210708 #182: Missinf P2p-device for proper wifi-direct. Can you help?

FT20016, 2025-06-20:
Dear developer(s),

I'm currently working on a project that requires establishing a Wi-Fi Direct group on a Raspberry Pi using your driver. 
While I’ve managed to initiate P2P groups using wpa_supplicant, I’ve encountered a blocking issue: the driver does not expose a p2p-device interface, which is crucial for proper interoperability with Android devices.

Specifically:
Android (13+) uses the WifiP2pManager API, which only discovers P2P peers that advertise themselves via a p2p-device interface.
Without this interface, Android cannot detect the Raspberry Pi, even if it's actively running p2p_find or has already created a GO group.
This behavior is confirmed by the iw dev output — the system lacks a separate p2p-dev-wlanX interface.
As a result, I cannot make the Pi act as a visible Group Owner (GO) in a stable Android↔Linux scenario, which breaks my application flow.

Could you kindly let me know:

if you confirm that p2p-device is not available
If support for p2p-device interface is on the roadmap
Whether there are technical blockers preventing its implementation
If you would consider enabling or accepting a patch for it
Any timeframe or workaround you could suggest

Your driver works well otherwise, and I’d really appreciate your feedback on this. Great work!

morrownr, 2025-06-20:
Hi @FT20016 

Hopefully I can provide enough information to help.

> if you confirm that p2p-device is not available

This driver is an old driver that is mostly based on the old WEXT set of APIs. It does not support P2P-Device. In fact, I am not aware of any Realtek chip based adapters or cards that use these old out-of-kernel drivers that do support P2P-device.

Thoughts: You have some options depending on what you are trying to do:

Option 1: Most of the Mediatek wifi adapters and cards do support P2P-device. Here is the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Reading menu items 1 and 2 can be very informative. Given the site that I maintain, I have a lot of USB WiFi adapters and am willing to check individual adapters and chips to check for P2P-device support. In fact, I just checked the mt7922 based card in my current system and it shows:

	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
		 * P2P-device

I am quite sure that all mt7921, mt7921k, mt7921au and mt7925 based adapters and cards support it also. I also think mt7912u and mt7910u based adapters support it. It depends on what you want and I am will to check if you want.

Option 2: I helped with the news drivers that were recently added to kernel 6.14 that support the rtl8821/11au and rtl8812au chips. The news drivers are part of the rtw88 series of drivers. If you are not at kernel 6.14 yet, you can install the driver from:

https://github.com/lwfinger/rtw88

Remember to fully uninstall the driver here with the following command:

$ sudo sh remove-driver.sh

If you see a problem with P2P-device support, which you probably will, you can make a report there and that stands a good chance of the support being added.

Good luck.

morrownr, 2025-06-22:
@FT20016 

Remember, I have adapters with almost all of the modern Mediatek chips so before you buy, you can ask me and I will test to make sure the chip/driver does P2P-device.

FT20016, 2025-06-22:
@morrownr

thank you!
Btw: do you know of any mediatek adapter that comes in nano form factor?
small size is important for my application.

Best,

Franco